nRF24/RF24Mesh #3: Data from far-away nodes are not routed to the master node

joycee12, 2015-01-12:
Hi!

**Example:** Let's say I have three nodes, namely Master, A and B.
* B is furthest from Master, where data cannot be sent directly to the Master.
* B is located close to A.
* A is closer to the Master, where the Master can still receive data from A.

> *W.r.t. http://tmrh20.github.io/RF24Mesh/index.html*
> "In the 'mesh' configuration sensors/nodes can move around physically, far from the 'master node' using other nodes to route traffic over extended distances."

I realised that the Master can only receive data from A, and not B. I tried to look at the classes for a method to use for this multi-hop function but I am unable to find it.

**May I know how should I go about doing the multi-hop?**

Thanks!

TMRh20, 2015-01-14:
Hello, FYI, when you open an issue on github, it sends me an email. Sending me additional direct emails is not going to speed up my response, quite the contrary.  

1. To answer your question, routing is automatic. If using RPi, try deleting dhcplist.txt.
2. This is experimental/developing open-source software. There is no how-to/technical support. (This area is for software issues, specific (mostly undocumented) questions, or suggestions)
3. See the [RF24Mesh](http://tmrh20.github.io/RF24Mesh/) and [RF24Network](http://tmrh20.github.io/RF24Network_Dev/) documentation for more info

closed 2015-01-17 by joycee12

Sharathmk99, 2016-10-08:
I'm also facing same problem. 
I'm using master as Arduino.
Please help

Sharathmk99, 2016-10-08:
@TMRh20  Please help..