nRF24/RF24Mesh #6: NOT AN ISSUE: dynamically NodeID

spaniakos, 2015-01-28:
hello TMRh20

i just made a script for dynamic allocation of NodeID using RPi as a master and the galileo as a client.
the basic consept is that i use NodeID = 255 as a master NodeID and i have 2 functions , one server and one client that handles a header message and allocated a Never Used so far Node.
The sketch requires reboot of the client, but as fas as i know it works. I dont have the chance to try it using multiple nodes, but if you are interested i can share my code for u to see (thow it is a basic consept),  but can help as an example for other to use (yet is not a material for the library i assume, but with some more testing it can be).

cheers
Spaniakos

TMRh20, 2015-01-28:
Definitely sounds interesting, I haven't really even considered dynamic nodeID allocation. I've been working more with RF24Mesh over the past few days, so this is good timing too. https://gist.github.com/ is an easy way to share unfinished code if you don't want to make a repository/fork/etc.

spaniakos, 2015-01-28:
nah its ok, i will push it to my fork if you library in 1-2 hours so you can see the example
i will push some notes as well (as i changed the define for the minimum time for setNodeID)

closed 2015-01-28 by spaniakos