nRF24/RF24Mesh #9: Question about meshing and payload type

xanto, 2015-04-25:
Hi, 
working with mesh library i have some issue with payload data.

For example with a payload of this type:

struct payload_st {
  float temperature;
  float humidity;
  float internal_temperature;
  float internal_humidity;
  int ivcref;
  
} data_to_send;

sometimes not received.

While no issue with this payload type:

struct payload_st {
  int temperature;
  int humidity;
  
} data_to_send;

I have read that can be a problem due to the lack of a capacitor on the hardware module. is a possible case?

Thanks

TMRh20, 2015-04-26:
Its possible, take a look at the wiki [here](https://github.com/TMRh20/RF24/wiki/GettingStarted-sketch-failing-to-send:-how-to-diagnose%3F)

Typically need more info on what exactly is happening though

xanto, 2015-04-27:
OK thanks.

Avamander, 2015-07-13:
Have you tried sending the data in multiple batches? (one float per message or two ints)

xanto, 2015-07-14:
I solved by inserting a 10uF capacitor  between VCC and GND on the module. Now works.

closed 2015-07-14 by xanto