nRF24/RF24Mesh #10: Examples how to send data from master to other nodes

Avamander, 2015-05-04:
Is there a way you could create a simple example how to send data from master to other nodes or from node to node? It took me quite a while to figure this out as a newbie (but I still don't know how to use RF24Mesh node ID's as destination instead of assigned RF24Network addresses)

ppotok, 2015-05-04:
I am interested in too.

TMRh20, 2015-05-05:
This requires not only a bit of time to create the examples, but there seems to be an issue with address lookups on nodes (except for master).

I've been thinking of adding some automated lookups, so you only have to send to specific nodeIDs when working with RF24Mesh, but it requires a bit of thought and planning. Most of my coding lately has been trying to work bugs out of the higher level libraries before looking at more features.

All in all RF24Mesh does need a lot of work, so feel free to leave this open to keep track of progress on this. I can't promise any timeframes, as warmer weather and sunny skies usually results in less development work, but we'll see how it goes.

Pidjey, 2015-05-12:
I am very interested in this, keep on it, you are doing a nice work

TMRh20, 2015-06-20:
FYI, I just pushed another commit that updates an example from the prev. commit, which included examples of sending node to node and performing nodeID lookups. The latest one also fixes some issues with data caching and address renewal. @Avamander I will also take a closer look at your commit showing sending from master to nodes as soon as I can.

Avamander, 2015-07-04:
I guess this is completed. I'm closing the issue.

closed 2015-07-05 by Avamander