nRF24/RF24Mesh #20: Adding this and other NRF24L01+ libraries to Arduino IDE's library manager.

Avamander, 2015-07-04:
It would be really nice if all NRF23L01(+) libraries would be available for download and update from the Arduino IDE's library manager.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

I and I guess other people would also really like to get the latest available versions of these libraries quickly from the IDE. At the moment updating these libraries is a bit confusing because if one download the libraries (RF24, RF24Mesh and network) directly from github (Download as an .zip) they do not work together, but using the .io webpage download links did not result in the latest version (if I remember correctly).

TMRh20, 2015-07-16:
Well the confusion between libraries should change over the next couple of months or so. RF24Network_DEV is the main holdup there, since its adds a lot of features from the base RF24Network by manicabug. I want to do a bit more work and testing on it, but will be moving it to master soon. RF24Mesh is probably nearly ready for a 'release' (once a timeout is added of course)

For now, I *think* it would be best to limit inclusion to just the core RF24 lib, since everything else is still mostly in development?

Avamander, 2015-07-16:
I think they can all be released, RF24Network and RF24MEsh can be marked as "0.9.0BETA" that way people will know it's beta.

sej7278, 2015-07-18:
as long as it can still be installed the "old fashioned way", as to me library manager is a joke, installing stuff in hidden directories etc; plus we still need 1.0.x support

Avamander, 2015-07-18:
Adding the library to the IDE's manager does not do anything else other
than that one can download/update the library from the IDE. Also,  the IDE
stores every library in either install folder or in the library folder in
sketchbook no hidden folders as the IDE does not hide anything.

Avamander, 2015-07-18:
Adding the library to the IDE's manager does not do anything else other
than that one can download/update the library from the IDE. Also,  the IDE
stores every library in either install folder or in the library folder in
sketchbook.

sej7278, 2015-07-18:
oh yes, unlike te awful Boards Manager, Library Manager seems to store things in the sketchbook directory, not ~/.arduino15/some/random/numbered/directory/

closed 2015-10-10 by Avamander