nRF24/RF24Mesh #23: Adding timeouts to all functions.

Avamander, 2015-07-13:
RF24Mesh should have timeout for every function that might just hang or fail. I really dislike the idea of a node just freezing silently so that there is nothing I could do via software to alert myself of it.

TMRh20, 2015-08-08:
I added a timeout to the address renewal. The downside to this is that it will allow an non-configured node to attempt sending etc. I haven't fully tested everything, but all the RF24Mesh functions should return/fail if the address is not renewed. The mesh.begin()  and renewAddress() functions will return true if OK.

Avamander, 2015-08-16:
Could you also add timeout to mesh.begin()?

TMRh20, 2015-08-17:
I've added the timeout to address renewal, but it will affect mesh.begin() by default. Or did you want to also be able to specify the timeout period like mesh.begin(channel, datarate, timeout) ??

Avamander, 2015-08-17:
That will do, although having the option to specify the timeout like that would be really awesome. It's your choice to add it or not. :)

closed 2016-02-14 by Avamander