nRF24/RF24Mesh #26: Questions about rf24/rf24network functions via rf24mesh

sej7278, 2015-07-19:
I'm about to move some sensors to using the RF24Mesh setup from RF24Network and was wondering how to access some of the old functions I use:

```
network.begin(90, 2);
radio.setRetries(15,15);
radio.setDataRate(RF24_250KBPS);
radio.setPALevel(RF24_PA_MAX);
radio.powerDown();
radio.powerUp();
```

can I just call them directly (not via the mesh instance) or are they not required anymore due to the self-healing nature of a mesh?

the powerUp/powerDown functions are very useful for battery setups, can they still be used, will there be side effects of cutting power to the radio e.g. will they have to rejoin the mesh every time or get a new DHCP address etc?

TMRh20, 2015-07-19:
The dataRate and the radio channel need to be called from the [mesh.begin()](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a34c28ab573f9f834331987c045f6d691) function because the mesh will call network.begin() to reconfigure itself, so needs to store the settings. The retry values are set automatically by the network, so it is generally recommended to use the [txTimeout](http://tmrh20.github.io/RF24Network/classRF24Network.html#ae6e8410fa817f1e13cb1cd90a96a2684) function of RF24Network to adjust it manually. 

For powerUp/powerDown, functionality it will mostly just work, but there are some things to consider. For example, say you have 1 master with 7 nodes. Four of those nodes end up being assigned as direct children of the master node. If all four of them go to sleep, without releasing their address, no other nodes will be able to renew their addresses. (The releaseAddress function may not work 100% of the time either, but should be relatively reliable)

There are a few ways to manage this, either by using 'static' rf24network nodes in conjunction with RF24Mesh nodes, of just releasing addresses on nodes that will be going to sleep. RF24Mesh does still need some work, and the class documentation needs a few updates, but its getting there.

sej7278, 2015-07-19:
thanks a lot for the update @TMRh20, i'll have a play when i get home and my new modules arrive. i've absolutely no problem with RF24Mesh being a bit beta, its all great as far as i'm concerned, RF24Network has served me well and I just fancy playing with mesh.

sej7278, 2015-09-08:
had a bit of a nightmare playing with this today, might be the arduino-side radio being fussy about the 3.3v power i'm feeding it, i must knock up a regulator circuit, the rpi-side is fine.

anyway, just about got the Pi & Arduino examples working, very intermittent and also only seems to work with my PA+LNA modules, the regular ones never seem to connect, the old RF24Network-only setup never had a problem.

i'm using master of all 3 libraries as it seems to have overtaken Development now.

any ideas?

Avamander, 2015-09-08:
@sej7278 The issue here might be that you have clone NRF modules. Clones and original chips tend to cause issues when you try to use them together. Try using two of the same kind together. The same problem has been discussed here https://github.com/TMRh20/RF24Mesh/issues/8.

I also suggest that you'd solder decoupling capacitors to the modules' 3.3V and ground pin.

TMRh20, 2015-09-08:
Hey, thanks for the input, although it is a bit discouraging. If you have a chance to take a look at the following issues, they seem like they might be related.  I've been meaning to take a closer look at this, but a series of hardware failures have pushed me back a bit.

https://github.com/TMRh20/RF24/issues/133 

https://github.com/TMRh20/RF24Mesh/issues/8

The issue looks like it may be something with the latest changes to RF24Network, since there seems to be a common trend of things working fine with the 'old' RF24Network. The main change is the use of dynamic payloads and dynamic acks, but dyn payloads appears to have been ruled out as the cause, since the pingpair_dyn example seems to work OK for the affected users. 

Another kicker is that my modules all seem to work OK. With my lna+pa modules, I can usually get away with using a 220uF cap + Arduino Mega 3.3V for simple tests, but its not that reliable. I have the [green modules](http://s2.electrodragon.com/wp-content/uploads/2011/11/2.4Ghz-Wireless-Nrf24l01-Module-Arduino-supported.jpg)  some [dark ones](http://thumbs3.ebaystatic.com/d/l225/m/m6Cn8jFnv6kJ7SekJ6TpesQ.jpg) and some [lna+pa modules](https://arduino-info.wikispaces.com/file/view/nRF24L01-LN-PA-1.jpg/243330387/nRF24L01-LN-PA-1.jpg) and they all seem to play well together.

I guess my biggest questions for anybody having this type of issue would be:

1. Does it work with RF24 pingpair_dyn example?
2. Does the gettingstarted.ino example work if radio.enableDynamicAck(); is called on both devices?
3. Does it work with current RF24Network rx & tx examples? 

Kind of stumped...

sej7278, 2015-09-08:
@Avamander yes i considered that, but i have at least 3 units i know aren't clones (and previously worked well)

sej7278, 2015-09-08:
@TMRh20 i don't mean to discourage you, your work is stellar, i was just reaching out as i didn't want to waste any more time on something that may be a known issue.

i'll try some more debugging in the morning. one thing to note - my old working setup doesn't seem to work when using the newer RF24/RF24Network, unfortunately i seem to have nuked the .git directories so its going to be hard to figure out what the working versions were.

i'm certainly going to try some caps and a regulator.

sej7278, 2015-09-08:
RF24 pingpair_dyn example doesn't work, here's the pi output:

```
RF24/examples/pingpair_dyn/
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xf0f0f0f03c 0xf0f0f0f05a
RX_ADDR_P2-5	 = 0x69 0x96 0xa5 0xc3
TX_ADDR		 = 0xcccccc3ecc
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x4c
RF_SETUP	 = 0x07
CONFIG		 = 0x0e
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for receiver, 1 for transmitter (CTRL+C to exit) 
>0
Role: Pong Back, awaiting transmission 
```

here's the arduino output (at the moment just an uno):

```
RF24/examples/pingpair_dyn/
ROLE: Ping out
Now sending length 4
RF24/examples/pingpair_dyn/
ROLE: Ping out
Now sending length 4
```

both devices have the same PA+LNA (and using latest master of all 3 libs)

TMRh20, 2015-09-08:
Hmmm, the addresses look odd on the RPi, but you may need to run the sketch a second time for them to show up correctly. The RPi may need a separate power supply for use with lna+pa modules, but there seems to be various info about the current capabiity of 3.3v on rpi. I'm assuming if you are using the same type of modules with no luck, it would be hardware related. If the modules are generally working with the core RF24 examples, then I would maybe test with different types of modules.

sej7278, 2015-09-08:
funnily enough i've just made a psu for the pi - takes from the 5v to an lm1117 and a couple of caps to give 3.3v, it works with my old sensors (but so did just taking 3.3v from the gpio).

i've gone though all (about 15 - 2 types of pa+lna and 2 types of black little ones) of my rf24 modules and all but one work fine with my old pi code and my old arduino sensors (also with the old RF24Network libs) so i'm really starting to think its in the newer RF24Network commits, but have no idea which ones.

the old working pi RF24 and RF24Network libs have a datestamp of "Sep 28  2014", but i have no way of telling what the arduino libs are as i've updated them without testing they work.

TMRh20, 2015-09-08:
Well please let me know what you find, and if you can possibly isolate the issues to a setting or combination of settings etc., that would solve the issue, or at least get us a lot closer. I was thinking earlier that it was dynamic payloads, so I added the option of ``` #define ENABLE_DYNAMIC_PAYLOADS ``` to RF24Network, but that seems to have been ruled out.

sej7278, 2015-09-08:
yup will do, maybe stepping through each commit (nice!)

sej7278, 2015-09-11:
i've noticed these two look a bit odd on the rpi with the new libs:

```
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
DYNPD/FEATURE	 = 0x00 0x00
```

with the old libs i get:

```
RX_ADDR_P0-1	 = 0xf0f0f0f03c 0xf0f0f0f05a
DYNPD/FEATURE    = 0x00 0x00
```

has the dynamic payload feature default changed now - i though it was supposed to be 0x00 for RF24Network and otherwise for the base RF24 lib?

TMRh20, 2015-09-11:
Yup, its enabled now in RF24Network, and should show as ``` DYNPD/FEATURE    = 0x3f 0x05 ```

I'm guessing you may have been using the [RF24Network Master-Testing branch](https://github.com/TMRh20/RF24Network/tree/master-TESTING) before, which is very similar to the archived version.

Maybe not clear before, but you can also revert to static payloads by commenting out the ``` #define ENABLE_DYNAMIC_PAYLOADS ``` in RF24Network_config.h

Info:
1. Dynamic payloads allow the network layer to present the user with the exact length of the received data (ie: a user payload type could be defined for handling variable length strings.. )
2. Dyn plds improve network throughput, but lowering the time required to communicate smaller payloads
3. Dyn plds can be corrupted, but (from my testing) only happens when throughput is maxed out, and only affects a small percentage of payloads.
4. Required also for many applications involving fragmentation/reassembly where the size if the payload is needed.

Dynamic-Ack is enabled also, to allow multicast/unicast functionality also. This is what I thought the problem would be related to at first, but am not so sure.

sej7278, 2015-09-11:
[fab8a65](https://github.com/TMRh20/RF24Network/commits/fab8a65) seems to be the last commit for my working setup.

with the new libs, commenting out the #define made no changes to what's shown by printDetails()

TMRh20, 2015-09-11:
Ok, the changes immediately following include https://github.com/TMRh20/RF24Network/commit/ad40fcb5c48694a701d82d143f4ca6a3fae1acee which introduces a new addressing scheme along with dynamicAcks, so I would think it would have something to do with that.

> with the new libs, commenting out the #define made no changes to what's shown by printDetails()

It should, but sometimes you have to force a total recompile for the changes to take effect ``` sudo make install -B ```

I'm probably more confused now though, were you still unable to get you modules working with the standard RF24 examples, or was that something else? If working, please re-test with dynamic payloads and dynamic ack per above or let me know the current results.

*The new network lib introduces many features, but limitations etc tend to make the implementation and usage a bit confusing...*

It also introduces network-acks, so if your code is handling a call-response type scenario, you will want use a non-acked payload type.

*User message types 1 through 64 will NOT be acknowledged by the network, while message types 65 through 127 will receive a network ACK.*

Example:

1. Node 011 sends a payload to master (00). No response is required, but node 01 wants to know if successful or not. Uses a payload type of 115 (char 's'), and the network confirms delivery.
2. Node 011 sends a payload to master (00), expecting a reply. Uses a payload type of  3  and the network does NOT confirm delivery. Node 00 (master) replies using a payload type of 116 (char 't') so the network will indicate success/fail properly.
3. Node 011 sends a payload to master (00), expecting a reply. Uses a payload type of  3  and the network does NOT confirm delivery. Node 00 (master) replies using the same payload type. Node 011 sends the final transmission using payload type 116 (char 't') so that delivery is acknowledged.

4. (INCORRECT) - Node 011 sends a payload to master (00). No response is required, but node 011 uses a payload type of 3. The network will not properly acknowledge payloads, and node 011 may think transmission was successful when not.
5. (INCORRECT) -  Node 011 sends a payload to master (00), expecting a reply. Uses a payload type of  115 (char 's')   and the network DOES confirm delivery. At the same time, node 00 is trying to send a reply to node 011, but the routing node (01) is busy confirming delivery.

TMRh20, 2015-09-11:
> it definitely seems dynamic payloads are enabled, as i have to pull the pi's power (or unplug the radio) to switch back to the old libs where they're disabled as per TMRh20/RF24Network#2

Hmm, when was the last time you updated the RF24 libs? This should be reset when begin() is called. I made a [nifty installer](http://tmrh20.github.io/RF24Installer/RPi/install.sh) for [RPi](http://tmrh20.github.io/RF24/RPi.html) so I can just delete and install fresh from github.

sej7278, 2015-09-11:
ah could be the new addressing then, as on both arduino and rpi i'm using a single digit int like:

```
const uint16_t this_node = 3;
const uint16_t other_node = 0;
```

it seems i should be using 00 (master) and then 01, 02, 03?

not using any sort of ack, the nodes only send to the master, the master never sends anything back.

where are the message types documented, i've not heard of them, well i spotted a reference to "M" header types in RF24Mesh, i'm just sending a char array.

i still can't get any examples working with the new libs (not even trying RF24Mesh anymore) but i've got to ensure i've got a clean environment i.e. ```sudo make install -B``` and unplug the radio.

sej7278, 2015-09-11:
yeah i tried the installer but that gets the latest libs of course, which don't work (yet) for me. the power/unplug fix was after the commit i'm stuck on.

TMRh20, 2015-09-11:
Yeah addresses should be specified in [octal](http://tmrh20.github.io/RF24Network/Addressing.html) but it only matters for numbers > 8 (I couldn't make this stuff up if I tried lol...)

Message types are user defined, so there is no reference, other than [here](http://tmrh20.github.io/RF24Network/structRF24NetworkHeader.html#ac99ab14e561670f4445d1fea7775e2db)  It doesn't really matter in this case - if a single transmission doesn't even work properly.

Sticking with the newest RF24 core lib should still work with the old RF24Network, but might make it easier to switch between them for testing.

sej7278, 2015-09-11:
ok, i'll give it all a go in the morning, i really want to get the latest libs working so i can start playing with mesh, but if i can at least get RF24 working it'll be easier. i'm hoping it is just the addressing.

the octal address reminded me i'm going to run out of toplevel node numbers soon, so guess i'll need to start using children like 011, 021, 031 etc.

thanks for all the help.

sej7278, 2015-09-12:
i tried helloworld_rx.cpp on the rpi and helloworld_tx.ino on the arduino, with 00/01 addresses and the newest libs and the arduino keeps saying "Sending...failed" but the pi says:

```
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3f
EN_RXADDR	 = 0x3f
RF_CH		 = 0x5a
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x05
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX
Received payload # 5 at 12000 
Received payload # 8 at 18000 
Received payload # 9 at 20000 
Received payload # 16 at 34000 
Received payload # 17 at 36000 
Received payload # 18 at 38000 
Received payload # 19 at 40000 
Received payload # 20 at 42000 
```

TMRh20, 2015-09-12:
Been doing a bit of reading, and from http://sigrok.org/wiki/Protocol_decoder:Nrf24l01

> A comment by a Nordic employee under a hackaday.com article titled "Nordic NRF24L01+ - real vs. fake" says that some clones interpret a flag different than the original chip, resulting in problems (quote: "When EN_DPL is activated, the NO_ACK bit get reversed in the real nRF-devices. They did such a good job of cloning they cloned the datasheet error into the device!!!"). The comment doesn't say which clone is meant, but from the article and the other comments it's most probably the SI24R1. 

So it sounds like the problem would only occur with the following conditions:
1. Dyn payloads are enabled
2. Dyn Ack is used
3. Some of the modules have the no-ack bit reversed...

That would mean that the only way to fix it is to somehow detect these modules, and flip the ack/no-ack options...

**Test Fix:**
On only one device (the suspected 'incompatible clone'), change line 1010  ``` radio.writeFast(frame_buffer, frame_size,multicast); ``` to ``` radio.writeFast(frame_buffer, frame_size,!multicast); ```

sej7278, 2015-09-12:
genius! that works, latest libs, just changed the arduino side.

so these are fakes, although i'm pretty sure they're not the SI24R1's, they weren't even around when i bought these, and i've tried 3-4 completely different types

TMRh20, 2015-09-13:
Interesting! I guess its safe to say the problem is definitely related to the ack functionality. 

With the change in software, I'm not 100% sure if it is actually addressing the issue, or just resolving the symptoms by using multicast. Since the problem is internal, I think the change would have to be applied to the devices using the clone/backwards modules, but even so, not sure it will work properly.

A simple way to test the fix is to apply the above change, and have only the tx device running. If no other devices are operational, and the transmissions are showing successful, then it is using multicast, and the 'fix' not really working. I'm guessing it has to be applied to the correct module, but that still may not work fully.

sej7278, 2015-09-13:
oddly enough i've just knocked up another board with the same type (and batch) of module and the fix doesn't work.

going back to the older RF24 and RF24Network it works fine.....

i'll try the test you mentioned, BRB

sej7278, 2015-09-13:
yes, TX only from the latest libs with the ! modification, and i get this on the board that doesn't work with the fix:

 ```
RF24Network/examples/helloworld_tx/
Sending...ok.
Sending...ok.
Sending...ok.
Sending...ok.
```

i'll try the board that does like the fix now

TMRh20, 2015-09-13:
> so its just blindly sending i guess, not checking (ack) that its really received?

Yeah, it looks like it. 

> BTW, i'm not using RF24Mesh, i thought that was the only lib that does multicast?

RF24Mesh is the only lib that *needs* multicast. RF24Network has multicast and unicast options built-in now, and RF24Mesh is basically just an overlay that adds some intelligence for address renewal/assignment/etc.

Also, if the above note about the ack being backwards during usage of dynamic payloads, it should have worked when RF24Network was recompiled, with ``` #define ENABLE_DYNAMIC_PAYLOADS ``` commented out. This only adds to my confusion... maybe it requires disabling both dynamic payloads and acks??? LOL...

RF24Mesh simply won't work without the dynamic-ack option, but it should work without dynamic payloads.

sej7278, 2015-09-13:
> maybe it requires disabling both dynamic payloads and acks

well that would take us back to the "old" RF24Network that i'm using wouldn't it? but make mesh not possible?

fixed payload size is not a problem for me, i tend to pad to the full 24-bits anyway, and i never use ACKs, probably why i've only just noticed this problem after a couple of years using the modules lol.

TMRh20, 2015-09-13:
I had a bit of a :bulb:  of using per-pipe ack/noack settings instead of using dynamic-ack for multicast/unicast, had some time, and did some testing with my modules and found some interesting results:

1. Had previously tested use with dynamic payloads disabled & it worked fine
2. Tested without dynamic ack (using autoAck(pipe,yes/no); instead) and it worked fine WITH dyn payloads
3. With the change to dyn_ack and dyn_payloads disabled, I had problems...
4. Fixed a bug with the unicast sendign to wrong pipe, and it worked OK with both dyn_ack and dyn_payloads disabled

Conclusion: My modules seemed to change their behaviour regarding ACKs when dyn_payloads and/or dyn_acks were enabled, depending on the combination. In [the latest DEV commit](https://github.com/TMRh20/RF24Network/commit/5b366aa9b62a57bb0ad59432ee8d5577bcad030c) I've removed dynamic acks (using per-pipe ack disable/enable) and disabled dynamic payloads. This leaves all of the current changes, including fragmentation/reassembly, the new addressing format, etc., without the use of dynamic payloads. 

After these changes, my modules now work with or without dyn_payloads enabled in RF24Network_config.h, and dyn_ack disabled, and is about as close to the 'old version' as possible, radio-feature-wise.

[Dev branch download](https://github.com/TMRh20/RF24Network/archive/Development.zip)

sej7278, 2015-09-13:
that's fixed it!!!!!

updated rpi and arduino to RF24Network (Development) without the ! fix, and now all my modules work, even without a capacitor/regulator from the Nano's crumby 3.3v pin!

TMRh20, 2015-09-13:
Awesome! :beers:  

I think we may have just discovered a way to test for these 'backwards' modules. It might even work now with dyn_payloads turned on, but *just working* is fine by me for now. I'll leave these changes in DEV for now, and look at merging into master.

sej7278, 2015-09-13:
ok shall i close the issue (or you can) or do you want to leave it open for reference?

TMRh20, 2015-09-13:
Well, if you aren't sick of this yet, could you run one more test with ``` #define ENABLE_DYNAMIC_PAYLOADS ``` uncommented in RF24Network_config.h with two nodes on your current working setup, and see if it still works?

Just to finalize things, if it causes compatibility issues with dyn_ack disabled, I will leave it disabled when merging into master I think.

sej7278, 2015-09-14:
yes, that works too with 3 different module types - both rpi and arduino latest Development libs with #define ENABLE_DYNAMIC_PAYLOADS uncommented:

```diff
   
     /** Enable dynamic payloads....**/
-    //#define ENABLE_DYNAMIC_PAYLOADS
+    #define ENABLE_DYNAMIC_PAYLOADS
 
     /** Debug Options */
```

so the problem was the ACK not the payload?

TMRh20, 2015-09-14:
> so the problem was the ACK not the payload?

Haha, I guess so, but only in combination with dynamic payloads I think!

> does this mean mesh will work, or does that need the ACKs?

Yeah, it still needs control of acks, but it should work ok using per-pipe acks instead of per-request (dynamic acks).

I guess I'd like to leave this open until I merge the changes into master.

TMRh20, 2015-09-23:
Thanks for all your help on this @sej7278 !

closed 2015-09-23 by TMRh20