nRF24/RF24Mesh #28: Adding support for modules similar to NRF24L01+

Avamander, 2015-07-22:
There are also modules named RFM73 and RFM70 that are almost exactly the same as nrf24l01+ chips, could you add support for these modules too when you are modifying the code to support the clones with broked dyn payloads?

TMRh20, 2015-08-01:
Possibly, I would have to look into the specific differences, but if somebody already has an idea what they are, that would help a lot. 

In RF24Network_Dev, there is a new option:  ``` #define ENABLE_DYNAMIC_PAYLOADS ``` which is available for testing of nodes that do not use dynamic payloads. Note: The network does not support *mixed* nodes. Either they all need to support dynamic_payloads, or all should have it disabled.

2bndy5, 2021-06-29:
This is entirely possible, but it should be disabled by default because there's A LOT of setup code needed which isn't universally the same between RFM70 and RFM73. We had a [prospective contribution that came close](https://github.com/nRF24/RF24/pull/754), but it violated the license of [the 3rd party library](https://github.com/jnk0le/RFM7x-lib) that the PR patched into a new example (not a native sustainable solution).