nRF24/RF24Mesh #29: Sleep doubt

turcoale, 2015-07-24:
Hi:

I make a red with RPI how master and Arduinos how childrens in a Mesh topology. The arduino nodes works with batery. I need sleep the nodes and send the data Each a Minute for power saving. My cuestion is:

if all nodes are asleep and wake up at different times, the mesh network is working?
As I can coordinate the time they wake up the mesh network to keep it running ?

thanks

Avamander, 2015-07-24:
If the node can not reach the Master you have to keep all nodes between it and the Master active during transmission. Mesh also needs to know when a Node goes to sleep, so do let the mesh know.

turcoale, 2015-07-24:
I can not understand how to correctly nodes are connected to mesh topology and also to sleep and wake up every so often to save batery

Avamander, 2015-07-24:
Check the documentation for futher information. It is all answered there.

closed 2015-12-14 by Avamander