nRF24/RF24Mesh #30: [Help] Extending the 4 node limit per circle

shyne99, 2015-07-28:
Hello,

As I understanded, where are limited to 4 Node per "circle" (or, in other terms, a node, including the master, cannot have more than 4 children).

I think that maybe there is a way to extand this limit. Can you explain us why this limit exist (I think that it come from somewhere...). Maybe this thread will gather people (including me) to work on a solution to expand it.

It's, for me, the main restriction of this incredible library.

TMRh20, 2015-07-28:
Network topology is pretty much statically defined by the manufacturer, in the [physical layer](https://en.wikipedia.org/wiki/OSI_model) and specified in the [datasheet](http://www.nordicsemi.com/jpn/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf) . If Enhanced ShockBurst (ESB) is disabled, multicasting can be used, but all of the features of ESB would be disabled and/or need to be replicated by software. Many changes would likely be needed at the network layer also. In my limited testing, multiple networks are able to function in close proximity using appropriately selected frequencies, so that seems to be a valid workaround in some cases.

SmartBlug, 2015-11-22:
Hello,
why 4 children and not 5 for MESH_MAX_CHILDREN ?

TMRh20, 2015-11-23:
@SmartBlug In a word, multicast.

In many words, the radios have 6 pipes available for listening, one for the parent node, four for the child nodes, and one designated for multicast. 

RF24Mesh uses the multicast addresses/pipes for polling, since many radios can receive and respond to a single request. The network is divided into multicast [levels](http://tmrh20.github.io/RF24Network/Tuning.html) , so a node will simply poll each level, looking for the strongest signal strength, then attach to the best available node at that level.

closed 2015-12-14 by Avamander