nRF24/RF24Mesh #32: Unable to upload RF24Mesh_Example.ino to ATTiny85

arjun-g, 2015-08-02:
Will RF24Mesh work in ATTiny85 ? I tried uploading the sketch but I am unable to do it. I removed all the Serial write. When I start building it just gets stuck at some point. Are any more steps needed for this ?

ppotok, 2015-08-02:
Do you know where it ends? if you use arduino ide turn on Debugging option in Preferences

arjun-g, 2015-08-02:
I tried with RF24Network_Dev and this is what I get.

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny85 -DF_CPU=8000000L -DARDUINO=10604 -DARDUINO_attiny -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Users\Arjun\AppData\Roaming\Arduino15\packages\attiny\hardware\avr\1.0.0\variants\tiny8 -IC:\Users\Arjun\Documents\Arduino\libraries\RF24 -IC:\Users\Arjun\Documents\Arduino\libraries\RF24Network -IC:\Users\Arjun\Documents\Arduino\libraries\RF24Mesh -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\Arjun\AppData\Local\Temp\build4861717368772641990.tmp\RF24Mesh_Example.cpp -o C:\Users\Arjun\AppData\Local\Temp\build4861717368772641990.tmp\RF24Mesh_Example.cpp.o 
In file included from RF24Mesh_Example.ino:12:0:
C:\Users\Arjun\Documents\Arduino\libraries\RF24Network/RF24Network.h:817:22: error: 'MAIN_BUFFER_SIZE' was not declared in this scope
  uint8_t frame_queue[MAIN_BUFFER_SIZE]; /**< Space for a small set of frames that need to be delivered to the app layer */
                      ^
In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:99:0,
                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:88,
                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:28,
                 from C:\Users\Arjun\Documents\Arduino\libraries\RF24/arch/ATTiny/RF24_arch_config.h:23,
                 from C:\Users\Arjun\Documents\Arduino\libraries\RF24/RF24_config.h:44,
                 from C:\Users\Arjun\Documents\Arduino\libraries\RF24/RF24.h:18,
                 from RF24Mesh_Example.ino:11:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h:146:27: error: 'MSTR' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                           ^
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h:146:65: error: 'DORD' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                                                                 ^
In file included from RF24Mesh_Example.ino:14:0:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h: At global scope:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI/SPI.h:156:7: error: redefinition of 'class SPIClass'
 class SPIClass {
       ^
In file included from C:\Users\Arjun\Documents\Arduino\libraries\RF24/arch/ATTiny/RF24_arch_config.h:28:0,
                 from C:\Users\Arjun\Documents\Arduino\libraries\RF24/RF24_config.h:44,
                 from C:\Users\Arjun\Documents\Arduino\libraries\RF24/RF24.h:18,
                 from RF24Mesh_Example.ino:11:
C:\Users\Arjun\Documents\Arduino\libraries\RF24/arch/ATTiny/spi.h:33:7: error: previous definition of 'class SPIClass'
 class SPIClass {
       ^
Error compiling.

arjun-g, 2015-08-02:
My master node would be a Arduino Uno I don't face any issue in uploading the master sketch in it. With ATTiny85 I am getting this.

arjun-g, 2015-08-02:
This is the modifed code I use

/** RF24Mesh_Example.ino by TMRh20
 *
 * This example sketch shows how to manually configure a node via RF24Mesh, and send data to the
 * master node.
 * The nodes will refresh their network address as soon as a single write fails. This allows the
 * nodes to change position in relation to each other and the master node.
 */

#include "RF24.h"
#include "RF24Network.h"
#include "RF24Mesh.h"
#include <SPI.h>

/**** Configure the nrf24l01 CE and CS pins ****/
RF24 radio(3, 3);
RF24Network network(radio);
RF24Mesh mesh(radio, network);

/**
 * User Configuration: nodeID - A unique identifier for each radio. Allows addressing
 * to change dynamically with physical changes to the mesh.
 *
 * In this example, configuration takes place below, prior to uploading the sketch to the device
 * A unique value from 1-255 must be configured for each node.
 * This will be stored in EEPROM on AVR devices, so remains persistent between further uploads, loss of power, etc.
 *
 **/
#define nodeID 1

uint32_t displayTimer = 0;

struct payload_t {
  unsigned long ms;
  unsigned long counter;
};

void setup() {
  // Set the nodeID manually
  mesh.setNodeID(nodeID);
  // Connect to the mesh
  mesh.begin();
}

void loop() {

  mesh.update();

  // Send to the master node every second
  if (millis() - displayTimer >= 1000) {
    displayTimer = millis();

    // Send an 'M' type message containing the current millis()
    if (!mesh.write(&displayTimer, 'M', sizeof(displayTimer))) {

      // If a write fails, check connectivity to the mesh network
      if ( ! mesh.checkConnection() ) {
        mesh.renewAddress();
      }
    }
  }

  while (network.available()) {
    RF24NetworkHeader header;
    payload_t payload;
    network.read(header, &payload, sizeof(payload));
  }
}

arjun-g, 2015-08-02:
Tried the above change and the first error is resolved. But the SPI issue is still there. Any idea on why its happening ?

TMRh20, 2015-08-02:
Yup, just don't ``` #include <SPI.h> ``` with ATtiny, the functions are built-in.

arjun-g, 2015-08-02:
Just found that. Thank you :)

arjun-g, 2015-08-02:
I am using the 3 pin configuration in ATTiny85. Arduino Uno will be the master. Will mesh work in this scenario ?

TMRh20, 2015-08-08:
If you haven't tried it yet, it should work.

Avamander, 2015-09-08:
@TMRh20 Close this issue please.

closed 2015-09-08 by TMRh20

venus696, 2018-06-30:
I am running an example from trmh20's library "Mesh Example Master To  Nodes".
I am getting the error mentioned below.
can any body help me to solve this.

Arduino: 1.8.5 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

In file included from C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.h:40:0,

                 from C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:5:

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp: In member function 'bool RF24Mesh::begin(uint8_t, rf24_datarate_e, uint32_t)':

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

 #define MESH_DEFAULT_ADDRESS NETWORK_DEFAULT_ADDRESS

                              ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:25:20: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

     mesh_address = MESH_DEFAULT_ADDRESS;

                    ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp: In member function 'uint8_t RF24Mesh::update()':

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

 #define MESH_DEFAULT_ADDRESS NETWORK_DEFAULT_ADDRESS

                              ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:49:24: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

     if(mesh_address == MESH_DEFAULT_ADDRESS){ return type; }

                        ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp: In member function 'bool RF24Mesh::write(uint16_t, const void*, uint8_t, size_t)':

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

 #define MESH_DEFAULT_ADDRESS NETWORK_DEFAULT_ADDRESS

                              ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:95:24: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

     if(mesh_address == MESH_DEFAULT_ADDRESS){ return 0; }

                        ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp: In member function 'bool RF24Mesh::write(const void*, uint8_t, size_t, uint8_t)':

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

 #define MESH_DEFAULT_ADDRESS NETWORK_DEFAULT_ADDRESS

                              ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:103:22: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

   if(mesh_address == MESH_DEFAULT_ADDRESS){ return 0; }

                      ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp: In member function 'bool RF24Mesh::checkConnection()':

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

[... 111 lines cut ...]

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:595:101: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

         if(  (addrList[i].address == newAddress && addrList[i].nodeID != from_id ) || newAddress == MESH_DEFAULT_ADDRESS){

                                                                                                     ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh_config.h:23:30: error: 'NETWORK_DEFAULT_ADDRESS' was not declared in this scope

 #define MESH_DEFAULT_ADDRESS NETWORK_DEFAULT_ADDRESS

                              ^

C:\Users\Venus Bilandi\Documents\Arduino\libraries\RF24Mesh-master\RF24Mesh.cpp:606:32: note: in expansion of macro 'MESH_DEFAULT_ADDRESS'

         if(header.from_node != MESH_DEFAULT_ADDRESS){ //Is NOT node 01 to 05

                                ^

exit status 1
Error compiling for board Arduino Pro or Pro Mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.