nRF24/RF24Mesh #34: Rf24audio through RF24mesh

ricmail85, 2015-08-29:
Dear TMRh20,
first of all thank you for your well done job, I really like your project!

I'm learning howto use these nrf24L01+ modules with my arduino. I'm mainly interested in the RF24mesh, because I'd like to do a sensor network comunicating with the master node.
I'd like to know if it is possible to use the RF24audio feature through such a mesh. I have a microphone mounted on a node and I want to listen on the master node. The point is that I want to use the mesh because the microphone node can be far away from the master, so I'd like that another node in the middle could route the audio traffic.

is it possible?
Thanks!

TMRh20, 2015-08-29:
Hi, thanks for your input.

Audio streaming is typically only possible with these modules in a 1-to-1 configuration, using the core library driver. This can be 'cheated' a bit by using multicasting, and having multiple radios listen to the same adderess.

nrf24l01 modules are half-duplex and operate up to 1-2Mbps in a 1-to-1 config. As soon as multiple devices are introduced, and because of the half-duplex nature of the system, communication speeds start to degrade quite rapidly.

Test Results using TCP/IP and RPi over RF24Network/RF24Mesh:
1. Max Speed (Direct node-to-node) - 20KB/s
2. Max Speed (1 relay node) - 8-10KB/s
3. Max Speed (2 relay nodes) - 4-5KB/s

The 'max speeds' listed are generally tested with only one node actively communicating, and throughput will drop if multiple devices are communicating.

For 8Khz audio, a stable stream @ 8KB/s is needed, so there is a chance it might be possible, but unfortunately RF24Audio is not directly compatible with RF24Mesh.

One option may be to set up a custom relay node, which could be used to extend the range of RF24Audio, but it would either have to be separate from the mesh, or possibly disconnect from the mesh and act as an audio relay for a set period. With a purely multicast relay, it may be possible.

ricmail85, 2015-08-29:
Ok, thank you for the useful explanation.

Therefore in a possible newtork made by several nodes and one master, I have to load the RF24audio on all the nodes/master in order to "listen". Then ckeck with them if there is an audio "sender" device. If the sender is seen by the master, ok. Otherwise, if a node founds an audio "sender" (not detected by the master) it redirect its output to the master. I think I can do both operation in the loop for, right?

thanks again.

TMRh20, 2015-08-29:
Well, not quite that easy. 

tl;dr: "I think it's much more complex than that."

Audio devices need to be separate from the mesh, and the first thing to do is probably test to see if a relay will even work along with RF24Audio. Initially, this would require 3 devices separate from the mesh entirely. If interested in testing a simple relay with RF24Audio, I can probably create a sketch to do so within in a day or two, otherwise it gets added to the list lol...

If it is possible to use a relay, then the RF24Audio lib would probably require some additions/changes to include an example and possibly relay options or something... ie: It would require some doing...

ricmail85, 2015-08-29:
Ok, thanks again for the reply. For me it can be very useful such a kind of sketch... But of course I have to better understand how the two libraries work.
I'll let you know about progress.

Another thing regards the RF24Mesh. In order to avoid the manual assignment of the node address (via serial, too) I wrote the piece of code below. It can help me in order to autosetup all nodes.
Do you think it is able to work properly?

Thanks!

--------------------
const int nodeID=AutoFindNodeID();

int AutoFindNodeID() {
  int maxID_=255, tmpID_=255;
  int nodeID_=mesh.getNodeID();

  if(nodeID_==0) {
    Serial.println("Autofind nodeID: ");
    //set the temporary one
    mesh.setNodeID(tmpID_);
    int n_addr=mesh.addrListTop;
    
    //now look for an available address (NB: now 255 is assigned, we have 1->254 possibilities)
    //start from 1, nodeID=0 is the master!
    for (int i = 1; i <= maxID_; i++) {
      int count=0;
      for (int k = 0; i < n_addr; i++) {
        if(mesh.addrList[k].nodeID!=i) {
          //all the nodeIDs must be different from i
          count=count+1;
        }
      }
      if(count==n_addr) {
        //i is ok as nodeID, take it
        nodeID_=i;
        break;
      }
    }
  } else {
    Serial.println("Loaded nodeID: ");
  }
  Serial.println(nodeID_);
  return nodeID_; 
}

TMRh20, 2015-08-30:
Well the thing with automatic nodeID assignment, is they are like IP addresses. If you assign them automatically, there is no way to identify which node is which. This may be OK in some situations, so I think something like that could work.

ricmail85, 2015-08-30:
Ok thanks again.

Last question: in my mesh, can I use a raspberry on the master and arduino on nodes?

bye!

TMRh20, 2015-08-30:
Yup, you can use RF24Mesh or RF24Gateway on the RPi.

ricmail85, 2015-09-06:
Thanks you for your great work!

bye!

closed 2015-10-18 by Avamander

ricmail85, 2015-11-11:
Hi, 

Regarding the possibility to transmit audio through mesh, what do you think is the best solution? Send an audio buffer as a payload or implement in some ways the rf24audio lib?

Thanks,
Bye.

TMRh20, 2015-11-11:
I would say it depends on your requirements. RF24Audio is still a bit limited in audio quality etc, but if you used devices with PA+LNA module + Arduino, you could have standalone audio devices that are simply triggered by a connected RF24Network/Mesh node.

The other side of the coin is transmitting data over the network, as payload data, which will reduce the data-rate significantly, and make it much harder if not impossible to achieve real time streaming with raw audio data. Include some audio compression and it might work, but I'm not sure.