nRF24/RF24Mesh #35: Problem on Arduino Mega

ricmail85, 2015-09-26:
Dear TMRh20,

I'm trying to setup a mesh network with 2 arduinos, one nano and one mega. By loading the examples on the nano I don't have problems. On the mega, after loading, the program hold on the mesh.update line... The mesh.begin is ok, but it does not goes beyond mesh.update... 

Do you know why? I tried changing the radios but it is the same... It seems related to the arduino MEGA board... But I'm sure it works because I used a lot.

bye!

Avamander, 2015-09-26:
Check your wiring and DEFINITELY solder a capacitor to the 3.3V and GND pins on the radio, Mega's 3.3V regulator is quite crappy. Report the results after doing so.

ricmail85, 2015-09-26:
Hi, thanks for the reply. I tried adding a 10 uF capacitor (I have this one at the moment) but the behavior is the same... sketch stucks at mesh.update() on the mega...

Avamander, 2015-09-26:
Check wiring. Attach bigger cap.

ricmail85, 2015-09-26:
Ok, I'll give it a try later and I'll let you know.
thanks again

ricmail85, 2015-09-26:
Ok, the problem was I didn't know that the MEGA has different SPI pins. I solved setting on the MEGA the pins as: CE(40), CSN(53), MOSI(51), MISO(50), SCK(52)
while on the nano I followed the usual ones: CE(7), CSN(8), MOSI(11), MISO(12), SCK(13)

bye!

closed 2015-09-27 by TMRh20