nRF24/RF24Mesh #37: RF24mesh network optimization Problem

ibrahimisim, 2015-10-05:
Hello ,

I setup  sensor network with one RPi (master node) and  six arduino pro mini (sensor node , slave) . when I run all devicess  I am waiting  about 10 minutes after that I can see all nodes on Rpi . in addition I connect relay to one arduino pro mini  when I send command to arduino I receive some time error and rarely I receive ok. Network works very slow. When I list node on Rpi  some devices change mesh addresses. Pls help me ! What can I do ?

Thanks,
Best Regards,

Avamander, 2015-10-05:
Could you please post your code or are you using builtin examples?

ibrahimisim, 2015-10-05:
I am using building examples.  master node examples on Rpi and node examples on 6 pcs arduino pro mini.

Avamander, 2015-10-05:
I am not familiar with rPI but is there anything else running on it, hogging up resources?

ibrahimisim, 2015-10-06:
Rpi  cpu usage about %25-30 . Sensor addresses are changing some time. And long time I can not ping to sensors from Rpi but some time I can ping to sensor.

TMRh20, 2015-10-06:
The default timeout for address renewal is only about 3 seconds right now, so if your sensors are not able to renew in that time, they may remain disconnected from the mesh.

If this is the case, settting ``` mesh.renewAddress(60000); ``` will set the address renewal timeout to 60 seconds, which should result in a higher number of nodes remaining connected. (I think 60 seconds would make a good default, but I haven't had a chance to change it) The same timeout needs to be added to mesh.begin() also. 

It is normal for the RF24Network address to change as nodes renew addresses, but the RF24Mesh nodeID should remain the same.

ibrahimisim, 2015-10-06:
Ok. Thanks. I will try your suggestion.  For address changing all nodes fix at their location and all nodes are active  I think  nodes shouldnt change their address. If one node is down  some nodes related down node  can change their address.

ibrahimisim, 2015-10-09:
Hi,

I tried your suggestion but problem is  still going on.

Avamander, 2015-10-10:
Try a total wipe of software and reinstall.

ibrahimisim, 2015-10-27:
Hi,

I tried a total wipe of software and reinstall . And than I  ran  RF24Mesh_Ncurses_Master.cpp  file  on Raspberry pi and  I ran RF24Mesh_Example.ino on 4 arduino pro mini board.  But network operates slowly . I attached ncurses printscreen of results . Many time ping is fail and some time ping is ok. And network speed is very slow pls see attached picture   data rate : 0.51 kbps.

Thanks for your helping.

Best Regards.
[image]
[image]
[image]
[image]

TMRh20, 2015-10-30:
This might just be 'normal' operation to a degree, but RF24Mesh is still in testing etc. so I'm not completely sure.

With these examples, each node is sending an acknowledged payload every second, and the master is doing the same, so it demonstrates how cross-talk can degrade throughput in a half-duplex system.

High throughput without using some sort of protocol can be a bit difficult, since a retry mechanism etc is not implemented. This can be demonstrated/tested by using two Raspberry Pi with RF24Gateway, since it uses the Linux TCP/IP stack.

Notes regarding reducing failure rates:

1. Control the timing of transmissions/responses, and utilize network-acks accordingly:
  * In this scenario, the master is sending at timed intervals, and the nodes are all sending to master at timed intervals. If the master node sends a request, and the nodes only send data in response to receiving a request, data collisions will be reduced.
  * The default examples use an 'M' type message (decimal 77) which is an acknowledged header type. In the type of data exchange suggested above, the message sent from master does not need to be acknowledged if a response is expected.
2. RF24Mesh design assumes master->node or node->master communication. Node<->node communication is not as efficient, nor is simultaneous or un-managed master<->node comms.
3. Using a protocol like TCP/IP is probably the most effective method of improving throughput. Experimentation/Investigation of a custom protocol for RF24Mesh is/has been on the list...

closed 2016-02-21 by Avamander