nRF24/RF24Mesh #38: How to know if message was router?

catan85, 2015-10-10:
Is there a way to know if a message was routed or it comes directly from the node?

TMRh20, 2015-10-11:
Hmm, not directly. It is possible to check the RF24Network header info, and compare the addresses, which represent the physical topology of the network. ie: Check if message is from a parent or child node, otherwise it is routed.

There is a bit of info on the addressing format [here](http://tmrh20.github.io/RF24Network/Addressing.html) and a link at the bottom to more info on octal formatting etc.

catan85, 2015-10-12:
Oh ok I understand.
I tried right now but I cannot make it routing my packets, on the server I always see packets coming from node 3 or 4 directly, I never see something like 013 or 014.
Is it normal? I tried different locations of the node but it always reach the server or nothing

Avamander, 2015-10-12:
Look at the RF24Network address, not nodeID

catan85, 2015-10-12:
My NodeID actually are:
0: server
1: node (from_node always = 4)
4: node (from_node alway = 3)
I'm looking at: 

header.from_node     // this return 3 or 4 (never 013 or 014 why?)

If I look at nodeID like you said it correctly reads nodeID = 1 and 4.

TMRh20, 2015-10-13:
Nodes will generally tend to converge around the master node, so with only 2 nodes, you would have to move one node out of range of the master node.

If you just want to test routing etc, in the [RF24Mesh_config.h](http://tmrh20.github.io/RF24Mesh/RF24Mesh__config_8h.html) file, you can force nodes to always route traffic by limiting the number of child nodes per parent to one. Set MESH_MAX_CHILDREN to 1 on the master, recompile, and delete the dhcplist.txt file if using Linux/RPi. Be sure to change it back to 4 for normal operation.

Per the [topology page](http://tmrh20.github.io/RF24Network/Tuning.html) the nodes are arranged into *levels* with the master being 0. When establishing a connection, nodes always start at level 0, so will always tend towards the master node, ensuring that nodes are always available to route traffic, and take the place of nodes that move or go offline. With only 2 nodes, usually only one will route traffic, and only if it needs to.

You can also print out the [address list](http://tmrh20.github.io/RF24Mesh/structRF24Mesh_1_1addrListStruct.html) to view all nodeIDs and addresses. The RF24Gateway or RF24Mesh ncurses interfaces are handy for this if using Linux/RPi.

catan85, 2015-10-13:
As you suggested I tried to set the MESH_MAX_CHILDREN to 1 and what I suspected was confirmed: messages are not routed to the master.
The situation is the following:
1. node 1 succesfully send data
2. node 4 succesfully send data. 
3. I set the max_children to 1
4. Only node 1 send data; messages from node 4 doesn't come.

I write my schetch here:
http://pastie.org/10477956 is the master
http://pastie.org/10477958 is the node 1
http://pastie.org/10477960 is the node 4
Thanks for you support!!
This evening I will check what append directly on the node. I suspect that they loops in send,fail,renew

catan85, 2015-10-13:
On the nodes I use this function:
network.multicastLevel(04);
network.multicastRelay = true;
maybe it could affect packet routing?

catan85, 2015-10-14:
ok, now I'm able to route packets from node 1 to node 4 to node 0. Also node 4 to node 1 to node 0. The problem now is that if the node between goes down, the leaf node continue to send, renew but the server never receive packets. If I call mesh.begin on server the communication comes back.

Avamander, 2015-10-14:
Why are you doing the routing manually if you are using the RF24Mesh
library? Something sounds off here. Could you please actually tell us what
you're trying to achieve (do) so that we could help you.

Also if you want to make sure your mesh works your nodes have to check if
they're connected periodically and before sending, if not then try to
reconnect and you also have to call .update and .DHCP functions (quite
often).)

catan85, 2015-10-14:
No the routing is executed automatically but for some reason if the relay central node goes down the leaf node isn't able to reconnect to the mesh server. It try to send packets and fail, then renew address but renew is never accomplished.
In the nodes I already call mesh.update()
mesh.DHCP i think it is to be called only on the master. Am I right?

Avamander, 2015-10-14:
If you use the examples, do examples reconnect and resend automatically for you? (They do for me)

(mesh.DHCP should indeed be called only on the master as often as possible)

catan85, 2015-10-14:
I was trying with master2nodes sample but it doesn't compile on 
payload = {ctr%3,ctr};
with debugger error:
RF24Mesh_Example_Master_To_Nodes:87: error: initializer fails to determine size of '__c'
For that reason I wrote my version of master..

Avamander, 2015-10-14:
Try redownloading the library and updating your Arduino IDE to the latest version. It compiles fine for me with 1.6.5

TMRh20, 2015-10-14:
> The problem now is that if the node between goes down, the leaf node continue to send, renew but the server never receive packets.

The node going down should release its address. If MESH_MAX_CHILDREN is set to 1, the master node (really every node) can only have 1 active child. 

The [docs](tmrh20.github.io/RF24Mesh) explain a lot of this info, but RF24Mesh still needs a bunch of work.

>  master_2_nodes error:

Try changing line 87 to ``` Serial.println( network.write(header, &payload, sizeof(payload)) == 1 ? F("Send OK") : F("Send Fail")); ```

catan85, 2015-10-15:
>Try changing line 87 to Serial.println( network.write(header, &payload, sizeof(payload)) == 1 ? >F("Send OK") : F("Send Fail"));
Yep, that solved compilation problem.. this evening I'll try with original examples.
Thank you

catan85, 2015-10-15:
Ok, I tried with original examples and also with them the problem is the same:
if I set MESH_MAX_CHILDREN = 1, start the communication is all ok.
Then when I switch-off the central node, the leaf node and the master stop communicating:
the leaf report "renewing address", the master report "send fail".
If I switch-on the central node the communication restart automatically.
All libs was downloaded on 15 september 2015 from tmrh20 website

TMRh20, 2015-10-16:
@catan85 That is exactly the expected behaviour unless the relay/routing node [releases its address](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a70e5a2a2a393bc13d63c97a62bf311b1)  

MESH_MAX_CHILDREN of 1 means only one node can attach to master. With this config, any time a node goes offline, it needs to release its address successfully, so another node can take its place.

catan85, 2015-10-16:
Mmm ok, I understand...  Isn't there a way to Force the address release on the master?

catan85, 2015-10-16:
If the leaf node is offline, or switched off it can't release its address

Avamander, 2015-10-16:
You need to do that on the slave node before it goes down.

catan85, 2015-10-16:
Ok but it's not possible to know if the node is going to be switched off, or if it's going to be out of range..
On the master I could check if a node goes offline for a certain period of time, then release its address. But is there a way to do it on the master? Or to completely reinitalize the nodes list?

TMRh20, 2015-10-17:
@catan85 See the [documentation](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html) The address list can be modified, and the network address set to 0 for a given nodeID, freeing up the space for another node.

catan85, 2015-11-09:
I tried to force zero address but nothing change... 

I tried also RF24Ethernet wish to automagically solve my problem.. it's a fantastic lib but in the end I've also there the problem with mesh network.. When a node goes down it doesnt' come back untill I reset the server.. When the node is a leaf it is worse than that... Unfortunately for my system that's not sustainable.. I will go back to the original rf24network with fixed nodes...  :(   

Maybe have you solved something in the last release?

Avamander, 2015-11-09:
It seems really weird that you are encountering these issues. Could you please upload your entire code what you are using on your nodes? We will be able to help you more then.

catan85, 2015-11-09:
Now i'm using the rf24ethernet slip gateway and server samples... I only
adjusted ce and csn Pin to 9,10 ; the resto of the sample is without
modification.  When the server goes down I have to reset gateway,  then
server, then the Connection comes back

Avamander, 2015-11-09:
Server samples? Did you mean node examples?

catan85, 2015-11-09:
no, I mean the SLIP http server (that listen on 10.10.3.2) and the SLIP
gateway of the RF24Ethernet lib.
But I have the same problems that I have on master and nodes samples of
RF24Mesh library.

TMRh20, 2015-11-10:
> Maybe have you solved something in the last release?

There have been some improvements in the latest two releases regarding address releases and renewals.

What exactly is the problem you are having now? In reference to a previous comment, nodes generally do need to release their addresses if going offline. There is some room for error, as long as nodes will be returning to the mesh, otherwise the addresses should be zeroed, as specified above.

Avamander, 2015-11-10:
@catan85 

Use only RF24Mesh, just for testing:
* Load Master example to one node
* Load Node example to other nodes, each of them need their own NodeID.

**Do not modify anything except CE pin, CSN pin and nodeID.**

Keep them all online as long as you can. Then come back and see if you are receiving x amount of messages, x is equal to the amount of nodes you have.

ricmail85, 2015-11-19:
Hi catan85,
I solved the problem of address releasing on the master in this way:
1. The master tries to send a M message to the node
2. If after N tentatives the node doesnt respond then I force its address to 0 on the master (mesh.addrlist(k).address=0)

Bye!
3. Upto now it works well

closed 2015-12-15 by Avamander