nRF24/RF24Mesh #39: write error

koverg70, 2015-10-16:
I created a master (0) and one child node (1). The child node gets an address, and sends a payload (millis) to master. Master receives it, but on the childe the write returns false. Then the checkConnectivity returns false also. So it renews the address which is successful.

There must be some problem with the master ACK-ing the payload. But comminication seems to work both ways, because child gets the address OK. What should I do the debug this problem?

TMRh20, 2015-10-17:
Are you testing with the included examples?

If so, its usually best to work down the stack, so test with RF24Network examples, and if no luck, test with RF24 examples. The idea is to narrow down the trouble to the appropriate layer, or to a physical/hardware issue if it doesn't work properly at all.

koverg70, 2015-10-19:
The problem is at the radio layer for sure. The packet arrives but the auto acknowledgement not working. Unsure wether the receiveds doesn't sends back or the oridinal sender doesn't receives.

It is known that some chips are false and it is suggested (even in the network config header), that I should disable dynamic payloads. The problem is, that when I disable it mesh stops working, de child nodes cannot allocate addresses. (Of course I updated the code on each node.)

koverg70, 2015-10-19:
Have you tried the current RF24Mesh code with #define ENABLE_DYNAMIC_PAYLOADS commented out? Does it work for you?

TMRh20, 2015-10-19:
Sorry, I thought you were asking for my assistance. If you so adamantly refuse to cooperate, you must understand that must refuse any further attempts at helping you.

closed 2015-10-19 by TMRh20