nRF24/RF24Mesh #40: Sending long string over mesh

ricmail85, 2015-10-17:
Dear TMRh20,

I'm trying to send quite long strings (500-1000 characters or more) over a mesh network. 
First question is: can I send all the characters in one big payload? Which is maximum size supported by the mesh.write/read function?

Otherwise, is it better to send character by character? In this case, I have to avoid ACK (for faster communications) or I can let them enabled? When declaring the msg_type, which is the character to set in order to avoid ACK payload?

thanks.
Bye

TMRh20, 2015-10-17:
> First question is: can I send all the characters in one big payload? 

Yeah, per the [docs](http://tmrh20.github.io/RF24Network/) RF24Network supports fragmentation/reassembly.

> Which is maximum size supported by the mesh.write/read function?

Technically limited by memory, defined in RF24Network_config.h as ``` #define MAIN_BUFFER_SIZE 144 + 10 ``` . With RPi and RF24Gateway, it supports 1500 Byte payloads no problem, but uses the Linux TCP/IP stack to manage the data.

> Otherwise, is it better to send character by character?

Not unless using RF24Ethernet, which will buffer data accordingly, and handle retries, validation, etc. 

> In this case, I have to avoid ACK (for faster communications) or I can let them enabled? When declaring the msg_type, which is the character to set in order to avoid ACK payload?

See the [documentation](http://tmrh20.github.io/RF24Network/structRF24NetworkHeader.html)

ricmail85, 2015-10-18:
Thanks for the reply.

I was thinking to use rf24ethernet. But I would like to separate the ethernet between sensors from internet. 
In other words, is it possible on the master arduino node to talk with sensor using 10.0.0.x ip addresses while the user requests come from a client with 192.168.x address?

Because I want to avoid sensors using ip addresses from my router... I prefer to have a private network for sensors with the master routing any request from internet...

Thanks.
Bye

TMRh20, 2015-10-18:
As with everything lately, the docs etc all need more info and updates, but there are two main methods of interfacing with RF24Ethernet:
1. Using a PC capable of SPI, ( interfacing directly with RF24 ) like Raspberry Pi - Uses RF24Gateway
2. Using a PC capable of SLIP ( Serial Line Interface Protocol ) like any Linux device - Uses an Arduino and the included SLIP example
3. *Todo* Look at PPP support

In any case, the PC connected to or serving as the master node controls all networking traffic. It can be configured to block all IP traffic, route IP traffic as required, etc.

Using RF24Ethernet requires a general understanding of networking principles. All IP addresses are statically assigned, so you can configure your network any way your want, all this provides is a network interface, connected to the RF24 mesh network.

ricmail85, 2015-10-22:
Ok thanks for the useful info.
Bye!

closed 2015-10-30 by TMRh20