nRF24/RF24Mesh #41: Mesh with Static routing nodes and sleeping leaf nodes. Issues

IgorTheWeak, 2015-10-19:
So I was building a project, where I needed Mesh network.(The dynamical properties). My nodes are running on batteries. I managed to build system so, that nodes wake up few times per hour(for testing every other 8 seconds) and send data. I have two static nodes(routing), which are plugged in wall. Everything seemed to work correctly once the addresses are assigned, but at one point it started to look so, that assigning static address doesn't work. 
So I have few possible issues in mind for those problems:

1. I changed CPU clock to 1MHz(Changed the bootloader and fuse settings).It seems to work.
2. I understand that addresses are saved to EEPROM, tried different things erasing EEPROM but no results.
3. I set static addresses as shown in example(Master-static). For some time it seemed to work, but I am not sure. Now it behaves like I never set them. For me, this seems to be the real problem. Because when nodes wake up, they can only connect to static nodes and master, but if one static node is connected to other static, then all nodes can not connect. Some times they just don't connect even if it seems that they could(free addresses).
4. May be the sleeping is also the problem. When they wake up, they do the mesh.update and then sends the data. Then go to sleep. As I mentioned in some cases all works fine, but then at one point part of network fails or doesn't configure. 

When the network is set, it seems to work(in some cases), but when its new or changing and trying to assign all the addresses it does not work fully or at all.

P.S. If I disable sleeping, the system doesn't work also.

Avamander, 2015-10-19:
* You have to release the addresses when the node goes to sleep or offline.
* As far as I know EEPROM is no longer used.
* If you upload the examples (Master and Node ones) to every Arduino do they work then?
* How far are the nodes from each other? Are you sure one node does not have to deal with too many childs because there is a limit?

IgorTheWeak, 2015-10-19:
Thank You for your reply!

1. If I upload Master and Node examples it does work. But it seems that sometimes it takes very long time for the network to reconfigure, if the physical location of nodes is changed. Also I noticed, that if node is disconnected, then next node can't connect to the master  (testing with one master and 5 leaf nodes connecting one at the time(5th could't connect even then when all 4 are disconnected)).

2. As for distance. I tried network with 10 nodes. I had luck having them all  on one table(but spaced far enough to avoid interference). Then I spread them evenly along one room and hallway(30m). In this case the addresses where assigned when the nodes stood on the table, then I put them in place and everything worked, but when I changed the network(positions) everything broke apart, and since then I had this struggle.(This is with 2 static nodes)

3. I use static addresses because node can't wake up other node if it is sleeping(that is what I read from nrf24 literature(I hope I am wrong)), so I need nodes, that do not sleep. So, they need static addresses.

4. If I release addresses when going into sleep. It is the same case if I restart them or turn on/off.?  If it is true then that doesn't seem to solve the case. As for child limit, I set them evenly, so that they can connect with two static and master without problem. But sometime the static is connected to other static and then it just doesn't work. It seems that it remembers it's unwanted address and doesn't reconnect to master, because it didn't save the address for it. 

The weird part is that while I was building the program and testing along, It seemed to work. And also at this point it seems that at some certain cases it works. But now, stepping back and trying to find breaking point, it looks that it just didn't.

Avamander, 2015-10-19:
The problem here is that you are not releasing the addresses if they go offline the Master has to know that and it seems that static nodes (you really don't need static addresses to stop a node from sleeping) are also causing some issues.

Please switch all nodes to dynamic addresses and release all addresses before putting the node to sleep. Then report back if you are still having issues.

IgorTheWeak, 2015-10-19:
Thanks for your answers. 

About the static addresses. I need those two nodes to be static, because they are the only ones who are awake. Because other nodes are waking up non-synchronized. So the only node they see is the static. If I give it dynamic address, It may not be connected to master at some point.

Avamander, 2015-10-19:
Waking up nodes also see dynamic nodes. If you set it to dynamically acquire an address it behaves exactly the same as a static node except its RF24Network address (not RF24Mesh) stays the same.

But that does not matter at all, if you wish to communicate with any node you can just use their NodeID (the one you defined at beginning) to use as an address, you do not need to use RF24Network address at all.

TMRh20, 2015-10-19:
Also, you might want try setting a higher timeout value on the nodes when calling renewAddress(), as in ``` mesh.renewAddress(60000); ``` to ensure nodes are not being left with no address.

In testing, it seems that some recent changes to the RF24Network layer may be negatively impacting address renewal, so for now the best workaround is to call renewAddress with a longer timeout period.

> As for child limit, I set them evenly, so that they can connect with two static and master without problem.

If you are adjusting the max_chidren_node variable, that is mostly for testing purposes, and generally not needed when using RF24Mesh under normal circumstances.

ricmail85, 2015-10-23:
Related to sleeping nodes, there is any way to do not allow routing/bridging message from these nodes? I.e. avoid that some nodes can have any child?

Thanks.

TMRh20, 2015-10-24:
No 'official' way to prevent this, though I suppose it might be a feature to think about adding. You could modify RF24Network on specific nodes by commenting out [this line](https://github.com/TMRh20/RF24Network/blob/master/RF24Network.cpp#L242)

This will prevent a specific child node from responding to any network polls to begin with. A flag or boolean value could easily be added to control this behaviour.

IgorTheWeak, 2015-10-27:
Hi!

Thanks for your responses. I wanted to update current progress and status.

I set all nodes to dynamic addressing. Two of them are not sleeping and works as routing nodes. All other nodes are sleeping, and waking up for few miliseconds to send battery status or if event ocures. 

I have problems at some points, when network is set and I change the program of the master or other nodes. Then it starts to not function correctly. It seems that it keeps storing address information or something, that conflicts with system. I tried to erase eeprom and flash in different ways. Didn't work.

But, then I reprogrammed the bootloader, fuses, set everything fresh. Now everything works like a charm and the network response is fast. I can change the physical and logical place of any node, switch them on/off and it works. 

But, with previous bootloader setting and fuses it worked too, until one point.

P.S. I didn't change the renewAddress() delay or set the releaseaddress(). 
But, the mentioned idea of disabling leaf node child possibility is good, because some times some node connects trough sleeping one, and it doesn't work.

IgorTheWeak, 2015-10-27:
I did try renewAddress(60000) and releaseaddress() when it was not working correctly, but it didn't help.

Avamander, 2015-10-27:
Again, it functions incorrectly if the master runs out of a dresses when
you do not release adresses of nodes going offline when they are going to
be either turned off or reflashed. Also, in addition to releasing adresses
when going offline, try releasing and reconnecting to the mesh every few
minutes, that way everything should keep working even if you fiddle with
the master.

IgorTheWeak, 2015-10-30:
Ok thanks for suggestions. I noticed that after some time(4days) experimenting and reflashing some nodes, my system suddenly started behave weird. I guess it is because the master run out of addresses as you mentioned.
 
The weird behaviour: 
   Random unreal address assignments. (node connected to node with 4 allready signed nodes)
   Some nodes reach the master only sometimes and half of network doesn't work.

So, I have to release address when going to sleep. But every time when I wake up I have to renew the address right? 

Also this means that going out of range or just disconnecting from mesh(force shut down) makes the system unstable over the time?

Is there a way in master node, to just restart dhcp?

Thanks in advance!

TMRh20, 2015-10-30:
@IgorTheWeak Thanks a lot for your testing and input. I did some testing of the address release process, and it looks like that was most likely the cause of your troubles. It was reverting to the default address before even sending the release request, so releases were failing constantly.

> Also this means that going out of range or just disconnecting from mesh(force shut down) makes the system unstable over the time?

With addresses not releasing properly, the mesh would get 'broken' over time. The latest commit should fix the problem. Once it is picked up by Arduino library manager, it should show up as v1.0.1. The default renewal/connect timeout has also been raised to 1 minute from 3 seconds in the latest update.

The mesh.releaseAddress(); function also returns a boolean value, so you can use the following to verify whether an address has been released properly:
```
Serial.println("Release ");
Serial.println(mesh.releaseAddress() ? "OK" : "Fail");
```

*edit to add: 
> Is there a way in master node, to just restart dhcp?

You could rewrite all of the assigned addresses to 0 on the master node, but you also would need to ensure that all nodes renew their addresses. The easiest way is by having them verify their connection every 30 seconds or whatever, and just leave the master node offline for 30 seconds to 1 minute.

IgorTheWeak, 2015-11-04:
Great, thanks! 

Will definetaly try that. 

I have also one other question. When my node wakes up and it can not connect to mesh it consumes a lot of power.(16mA for few seconds) I wanted to decrease number of renewaddress() tries and I guess the renewaddress() delay. It is more important for my system to save power, than keep the network realtime. So I wondered is it ok to do that. I noticed, that usually it takes practicaly no time to connect to mesh, so if I wake up every few minutes, then it should connect to mesh in that same short period of time, if the network is ready for that. If not, it is ok to go again to sleep and try next time. I need my nodes, to be able to send sensor update once every half hour.

TMRh20, 2015-11-04:
Yup, use ```mesh.renewAddress(yourChosenDelayInMilliSeconds);```

IgorTheWeak, 2015-11-19:
Hi. I downloaded the new libraries from github and installed the new code with releaseAddress. It seems to be working. But I wanted to make sure how to do this correctly, because it seems that it isn't particulary ultra low power now. (sometimes it takes time to reconnect to mesh(few seconds is a lot for my purpose)).

But thanks anyway. 

Structure of my code:

Setup > 

setnodeID
mesh.begin 
mesh.update

Main > 
watchdogwakeup
radio.powerup
mesh.begin
mesh.update
mesh.sendmessiage
mesh.releaseaddress
radio.powerdown
gotosleep

IgorTheWeak, 2015-11-20:
Can I not do the mesh.update in the main loop for sleeping nodes? Because, they do not need to maintain the mesh, for my understanding. Also, I removed mesh.begin from main loop and replaced with renewaddress.

TMRh20, 2015-11-21:
There may be a few things that can be done to speed up/improve address renewal, but I haven't been able to get much done lately. The general flow should be as follows:

Setup >

1. SetNodeID
2. mesh.begin

Main:
1. watchdogwakeup
2. radio.powerup
3. mesh.update
4. mesh.sendmessiage
5. mesh.releaseaddress
6. radio.powerdown
7. gotosleep

> Can I not do the mesh.update in the main loop for sleeping nodes? Because, they do not need to maintain the mesh, for my understanding.

You typically need to call mesh.update() on every node, because it handles reception and routing. Technically, you probably could get away without calling it, in some cases, but if the radio buffer fills up, you won't be able to transmit either.

IgorTheWeak, 2015-11-24:
Shouldn't there be renewaddress after wakeup?  I tried the program flow as you described, but it doesn't send any data. Only shows that node was connected.

TMRh20, 2015-11-25:
Ooops, definitely should be renew after wake-up. You would think who wrote the lib would know how it works lol.

IgorTheWeak, 2015-12-01:
Hi!

I have been testing for some time now with this new code. Basically network works, but it takes quite sime time to renew address, which is bad for my low power purposes. Sometimes it even takes up to 10 -20 seconds to renew the address, and if I limit this delay, for example, 1 sec, then the network isn't responsive. Half of the time the nodes can't reach the master with its message. I noticed that sometimes the renewal takes a lot less than second, so theoretically it  can work. Also I noticed, that some nodes(2/8) crashes, entering infinite loop(after few hours of working), although I have watchdog timer(not sure, maybe my fault with code). 

P.S. The lib I used is about 10 days old. Saw that there is updates, will try them.

TMRh20, 2015-12-02:
I haven't been having any major problems with it myself, with nodes running reliably for many days at a time, using a few different versions of the software.

That being said, I have noticed some intermittent issues with address renewal that are harder to track down, so I added some changes and extra debug info just now. The main change is capturing up to 4 poll responses and requesting from each, along with slower timing. There are some minor related network changes I'm playing with but they are not tested/proven enough.  

I was going to work on other RF24Mesh stuff today, but you seemed to be having problems, & since it's officially Dec.... merry xmas lol.

IgorTheWeak, 2015-12-02:
Thanks and even merrier Christmas to You!

I downloaded the latest libraries rf24/network/mesh and it is working much smoother now. Haven't noticed any major problems yet. 

By the way. About my project. I am creating low power wireless sensor network with nodes, that can run on one 650mAh battery more than a year (with optimized network 2,5years). I have pir sensor and battery voltage monitoring as well temperature sensor. With present hardware it is consuming about 30uA while sleeping. And if the status message takes less than 200ms to send than it barely changes.

IgorTheWeak, 2015-12-02:
One of the possible reasons I thought of, is that master is doing a lot of things, so may be sometimes it takes longer to recieve message simply because it is processing something at that time. Although it cannot possibly be more than a second.

TMRh20, 2015-12-02:
> One of the possible reasons I thought of, is that master is doing a lot of things, so may be sometimes it takes longer to recieve message simply because it is processing something at that time.

That is a big part of it. Right now its a bit of a complex process due to the nature of the radio addressing etc. so there are a few steps involved in renewal: 

1. Broadcast a multicast NETWORK_POLL 
2. Wait for responses
3. Send a direct multicast write (address request) to a responding (intermediary) node
4. That node relays the message to master using a standard send
5. Master looks for free addresses based on where the request came from
6. Master sends a response to the intermediary node
7. Intermediary node sends a final multicast write to the initial node
8. Upon receipt of an address response, node sends a final address confirmation to master
9. If the confirmation was OK, node is active and ready, otherwise retry address renewal. 
10. Master adds node to the list.

There is a bunch of delays in there, and in simple testing/theory it seems they can be shortened, but in reality, it needs to allow for additional processing and delays of intermediate nodes. I've been thinking about a more decentralized system, where individual nodes handle assignment to children, but for a hobby project, it takes a lot of time. Currently, the best low-power solution is to use RF24Network and/or just use RF24Mesh for initial assignment, and don't release/renew addresses.

IgorTheWeak, 2015-12-07:
Thanks for explaning address renewal process.  For now it averages about 200ms which is ok. 

One other thing.
I changed my MCU clock to 2MHz(16/8), and I was wondering if there is something to change in library or in code. I understand that all of the delays in code need to be 8 times greater(renewaddress etc.). Would there be any other problems, because now I am not sure weather it is simple delay problem or the library is specificaly written for given frequencies.

I need lower clock frequency, because 16 and 8MHz ar unstable for my power supply(1.8-3V). I changed the necessary arduino files(board,bootloader).

TMRh20, 2015-12-09:
Not sure, but I don't think the slower clock speed makes a huge difference overall. It can make a noticeable difference in performance but you'd have to test for yourself.

closed 2015-12-14 by Avamander