nRF24/RF24Mesh #43: checks duplicates IDs

gepd, 2015-11-03:
There is a way to check if the ID assigned to a node is already in the network?

I was thinking would be a good idea to implement something like this in the begin function, it could return 2 when the ID is duplicated. I'm usign something like dinamics IDs with my nodes, so will be really usefull to have this function.

I'm starting with this library so I should read a lot of code to know it, as I undertand the mesh library works with NRF24, RF24Network and RF24Gateway.

Can someone guide me a little bit how could be possible to implement something like this?

Thanks!

TMRh20, 2015-11-04:
In looking at https://github.com/TMRh20/RF24Mesh/pull/7/files and in thinking about it more, it is probably a feature that could be added fairly easily. I have to review the code, but I think a given node could either use a blank or specific nodeID when performing an address request, in which case the master would just assign the first free nodeID to that node. Other than that it could be done manually, kind of, but integrating the feature in the library is probably the easiest.

gepd, 2015-11-04:
In my project the sensors always have a unique ID, but thinking now, all nodes can be dynamic.

So if a node can't send the data where the master node is located, I can put an **Extensor** node, it will take the data from **Node 1** (Sensor) and will transmit it to the **Master** (in case that the master is is in sight).

``| Sensor | >>>>>> | Extensor | >>>>>> | Master |``

 Until now, I set the sensor in the "search mode", this mode works sending one byte to every ID (in the range of 1 to 255, excluding its own id) so when it detects that the message was received correctly, it sets that node as destiny.

The extensor has the same "search mode" but it includes the Master, so first will try to find the master and if it's not available will continue with the rest of the range (1 to 255 excluding its own id and the source node)

My idea is to automate and simplify this process. Until now this/these library/ies seems to be a good option, and I'm interested to collaborate on anything to improve it.

The pull request has something with what I want but not at all.

I think, the first step to do this is to add the hability to know if the master already assigned the ID to other node, it's the only modification in the core of the library, the rest can be implemented in a sketch

TMRh20, 2015-11-05:
Hmm, I think you may be misunderstanding the nature of the mesh, which is fairly easy to do considering the lack of documentation etc. 

The mesh/network already handles routing, addressing etc. so it would be more efficient to simply renew the address, and attempt sending to master. If no luck, renew again, and retry, etc.

> I think, the first step to do this is to add the hability to know if the master already assigned the ID to other node, it's the only modification in the core of the library, the rest can be implemented in a sketch

Duplicate IDs should be avoided at all costs. If the master already assigned the ID to another node, you are potentially interfering with its communication. With fully dynamic nodeID + RF24Network_Address assignment, duplicates would be avoided.

gepd, 2015-11-05:
I understood almost everything. I know the mesh/network already handles routing and addressing.

My problem is with the IDs. What I'm trying to do is avoid duplicates. What I didn't understand is ``nodeID + RF24Network_Address `` I think this is what I'm trying to do, but you said already is possible to do it?

TMRh20, 2015-11-05:
Yes, RF24Network addresses are already dynamic, and can be looked at like MAC addresses. NodeIDs are more like IP addresses, and just like with a home network, if they are assigned dynamically, and change often, a specific host (like a server) will be difficult to identify without some sort of DNS type system.

NodeID + RF24Net. assignment would just require a few changes to RF24Mesh, and I can probably look at doing this right away. I was planning on getting in some work on this lib tomorrow.

gepd, 2015-11-05:
In my case the nodes won't change often, is just a way to simplify the process to add "extenders", in some case can be 1 in others 4.

Thanks TMRh20, I'll be waiting to test the changes.

TMRh20, 2015-11-05:
I'm not sure what you mean by 'adding extenders', since all routing etc. is automatic.

gepd, 2015-11-05:
In my project, the extenders are like the WiFi extender.

So what I'm trying to avoid is to change ID manually in each device to avoid duplicates IDs.

I know the addresses are automatically assigned. As I said, I'm trying to do the same with IDs

TMRh20, 2015-11-05:
In experimenting a bit and looking further at this today, dynamic nodeIDs are just not practical.

In order for the network to function, there must be a way to distinguish between nodes. This could be anything, the only requirement is that it is **unique**. 

I was thinking initially that this could be managed by using random numbers for each request, but that leaves the master without any way to manage orphaned nodeIDs/Addresses. A random number generated at startup would not work either, because 'random' is not unique. 

That leaves the only option for dynamic nodeID assignment as a sort of custom thing that would work like this:
1. Each new node would have to be in range of the master
2. A new function like getNodeID() would need to be made
3. The node would use multicasting (just like with address renewal) to request a nodeID without interfering with other nodes

Again, the problem here is that you still need some sort of unique identifier to identify nodes, or you end up with orphaned nodeIDs/Addresses, and the mesh fails.

This leads me to the conclusion that dynamic nodeIDs are simply not practical, and won't provide any benefit.

gepd, 2015-11-05:
I understand the result of your experiment.

I've other doubt, and maybe, it can be because I'm not undertstanding how the library works at all. 

Lets think in a mesh with 5 devices (nodes)

``| Node ID4 | ===== | Node ID3 | ===== | Node ID2 | ===== | Node ID1 | ===== | Master ID0 |``

- If only Node 1 is in the range of Master, this mesh only can works sending messages between the master and ID1. If the node 4 want to send a data to the Master it can't because, in spite of have and unique ID, isn't in the range, I'm right?

Maybe I'm misunderstanding something, can you explain me that?

TMRh20, 2015-11-06:
> If only Node 1 is in the range of Master, this mesh only can works sending messages between the master and ID1.

No, routing is automatic. That is the whole point of the mesh.

Any node that cannot contact the master will try to attach to another node that can. The 'other' node will relay messages automatically. See http://tmrh20.github.io/RF24Network/Tuning.html

gepd, 2015-11-06:
Thanks for all you explanation TMRh20 I was misunderstanding the concept of the mesh.

Just add something else, the best way to think in the IDs and addresses is like; ID = MAC, and addresses = IP. Thinking in that way, automatic IDs hasn't any sense.

*OFFTOPIC:*
I had not read that part of the documentation. It has a lot of information I didn't knew, I liked the feature "Dual Headed Operation" I found this [video](https://www.youtube.com/watch?v=gtM832Z0ujE&list=WL&index=24) who compares the ranges of different version of the nRF24l01 One of them can transmit up to 1600mt, but it is only a transmitter with "Dual Headed Operation" you can solve the problem very easy, using one normal module as receiver, and the longer range as transmitter using the same library.

closed 2015-11-06 by gepd

mamama1, 2017-01-03:
Hi

I'd like to hook in here, although the issue is closed over a year now.
I'm currently developing a system using your mesh library which uses many sensor/switch/dimmer/etc. nodes + one controlling master node.

each (master) node will have its own, unique serial number (provided by ATECC508A or similar). node ID assignment has to be semi-dynamic here because in case I deploy this as a product, the node IDs can't be set uniquely to the devices prior to shipment. if I'd do that, I'd only be able to deploy 255 devices, as there aren't more unique IDs ;-)

so i was going to implement some kind of "pairing" process, where all new nodes have to be paired once with the master node, which in turn saves the node id + node serial number in its EEPROM. in turn, the client node saves the master nodes serial number + its assigned node id in its EEPROM.

i was going to use node IDs 1-9 as "reserved" node IDs, and ID 1 should be the one each new node uses for initial connection and pairing. (which in turn means that only one new node at a time can be paired with the master node). after pairing and setting the new node ID, the client node would restart and use the assigned node id further on until it is factoryresetted by a button or something (which would turn the ID back to 1).

what would be your approach to implement this, is there a chance you'll implement something in your library that would help me with my challenge?

what happens if there are two non-unique node IDs on the network? is there a way for the master node to tell the non-unique node to go away or is there a way for the non-unique node do detect that theres already another node with the same id?

just to clarify: I'm not a company, just a hobbyist who tries to develop something that COULD be a product sometime.

thanks!

Avamander, 2017-01-03:
Have you thought about using two MCUs on separate channels that in the end
merge their processed data stream to be interpreted by something else?
You'd then have 255 times MCU count on channels and you will avoid
congestion and can pipe through a lot more data.

mamama1, 2017-01-03:
Thanks for your reply!

I'm sorry maybe I didn't poit that out clear enogh. it is not a problem that one master node will only be able to communicate with 255 client nodes.
the problem is, that i have to implement dynamic node id assignment because I don't want to hardcode the node IDs to the nodes. also hardware dip switches or similar are not possible in this scenario.
as described, some pairing process will have to happen where the node id gets assigned by the root node and saved by both sides (client and root/master).
unique serial number will be 72 bit so thats a bit much to replace the node id which currently fits within one single byte.

Avamander, 2017-01-03:
Use placeholder NodeID, negotiate, restart with new, permanent nodeID.

mamama1, 2017-01-03:
thats what i described in my first post. my concern was, if there is any way to detect duplicate node ids either on client or master side.
if two new devices with the placeholder id come online, i dont want to get unpredictable results. so some mechanism to request a node id without actually having one yet would be cool.

Avamander, 2017-01-03:
Unless you modify the library already and make the node ask before
connecting if there already is a node with that ID.

mamama1, 2017-01-03:
thats exactly what i was asking help for :-) is this the preferred way? maybe tmrh20 has another approach? should i implement it and send a pull request? i'm far away from tmrh20's coding knowledge so i guess my implementation wouldn't be by far as good as his.

Avamander, 2017-01-03:
You can try and implement that, submit a pull request and we can work on
making it good together.

mamama1, 2017-01-03:
ok thanks ill try and submit later!

mamama1, 2017-01-04:
ok i just looked into the code and realized that the only clean solution would be to replace the node id (which is now unique) with some other, much bigger ID which is guaranteed to be unique (like the 72 bit serial number of the ATECC508A crypto IC).

thats because the node ID is kind of the "MAC address" and if you have two devices on your network which use the same mac address, there is no way to distinguish between them.

the only way would be to send a request to a node ID and wait how many replies you get. again, then there would not be any way to tell only ONE of the nodes to change its address. both would receive the request and both would change the ID and probably end up again with the same ID.

changing the unique node id from char to like char[9] require a huge amount of changes within the library + it would increase memory demands and execution speed as my devices (Atmel 328p and 2560) are both 8 bit only.

I'm not sure if I should implement this and submit a pull request or if I rather should fork the Mesh library (and maybe even underlying libraries if necessary).

This is something which would push the library a bit into the "professional" direction i guess, on the other hand I think it is questionable whether this will be practical for most of the users as they will need some external unique id provider (like ATECC508A) or define a much larger unique ID at the beginning of their program than they do now.

Avamander, 2017-01-04:
One way would be keeping the MAC address separate from nodeID. Then once in
a while request their MAC by targeting certain NodeID, when you get two
replies with separate MACs you can tell again that nodeID(two nodes)
something similar to "this [MAC] change nodeID and restart".

mamama1, 2017-01-04:
Hi

i tried to implement the unique 9 byte ID within the network header. i've corrected all the functions within the rf24network library where memcopy operations and similar were done with the hardcoded sizes of the old header (8 bytes header + 2 bytes frame_size) and i also corrected the MAIN_BUFFER_SIZE and the MAX_PAYLOAD_SIZE. headers are now 19 bytes long instead of 10 (17 + 2 instead of 8 + 2).

normal communication seems to work ok but when it comes to fragmentation, the bridge stops responding to requests. the last output i see from the enabled debugging output is:

> 12974: MAC Received on 5 id 21552 from 05 to 00 type 148
> 12975: NET message 3130
> 12976: NET Enqueue @0
> 12979: MAC Received on 5 id 21552 from 05 to 00 type 149
> 12982: NET message 3635
> 12984: NET Enqueue @0
> 12987: MAC Received on 5 id 21552 from 05 to 00 type 149
> 12991: NET message 3130
> 12993: NET Enqueue @0

so it seems that one or more packets (at least the last packet with type 150 which is NETWORK_LAST_FRAGMENT) don't get processed. do you know where to alter the libary/libraries to increase the header size by a 9 byte char array?

i was looking for the error for hours now and can't find a solution.

thanks!

mamama1, 2017-01-05:
hi

so after more hours of debugging ä, searching and eventually destroying the library completely i've decided to start over tomorrow or on saturday.

would you be willing to help me with this? as this is mainly affecting the RF24Network library, should I make the changes which I think are necessary and submit a pull request there, even if I know that it wouldn't work when it comes to fragmentation? At least then you'd see what I've changed...

Basically for me it would be ideal to entirely replace the node id with the 72 bit serial number but i guess that would involve even more messing around + if we don't replace the node id but add the serial number to the header we could make it configurable whether to use it or not (i guess not everyone would want to use that 72 bit hog).

ps I added the serial number char array to the end of the header struct when i was trying to implement it.

pps: in the rf24network.h it says that the node id is like the ip address and the network address which gets assigned is to be seen like the mac address. i think i understand why this has been written like this but in my point of view, the mac address is the node id as it is fixed and unique and the network address is the ip as it is dynamic and gets assigned by the master node... am i wrong here?

thanks for your effort and help!

mamama1, 2017-01-05:
you mean also without adding the 9 byte serial number to the header?

Avamander, 2017-01-05:
Indeed.

Avamander, 2017-01-05:
Indeed, but you can exchange the defined unique MAC only when negotiating
connection too, you shouldn't have the need to send the unique MAC once the
unique nodeID has been acquired (and saved).

mamama1, 2017-01-06:
i'm sorry i'm feeling a bit stupid at the moment. i don't really understand your approach... could you please be a bit more detailled? please try to explain it for orang utans... :-)

Avamander, 2017-01-06:
You have to initate a connection to the master of the mesh, right? During
that the master gets to know what's the node's nodeID and the node get's an
RF24Network address. Now, instead of sending the master nodeID during
negotiation send unique MAC instead, now the master should return both
RF24Network address and nodeID.

mamama1, 2017-01-07:
how gets the master to know to whom he has to send the nodeid/rf24network address response to?
the request package already has to contain a node id or not?

mamama1, 2017-01-07:
ok i do see now that the client node polls the network first and then sends a NETWORK_REQ_ADDRESS message. in the header of this NETWORK_REQ_ADDRESS message, the "from_node" field contains 2340.
this is the MESH_DEFAULT_ADDRESS (octal 04444).

so when two nodes at the same time request an address, they'll maybe end up getting the same RF24Network address, or did I miss something?

two nodes send a request, the master sends a response and both nodes accept and confirm it.

if we compare to real DHCP, the NIC always has its MAC address and the DHCP request, offer and response always contain the unique MAC address.