nRF24/RF24Mesh #44: Mesh node list not available on nodes other than master

ricmail85, 2015-11-14:
Hi,

I'm building a mesh with a master and two nodes. If I check the node's list on the master it correctly displays the two node IDs and addresses.
If I try to see the mesh nodelist on the node side (using mesh.addrList[k].nodeID) I have these problems:

1. mesh.addrListTop is 0, while if I do sizeof(mesh.addrList) the result is 2 (as it should be).
2. If I check the nodeIDs on the node, it displays 0 (the master) and 80, while the other node has a node id 1...

why?
thanks.

bye!

TMRh20, 2015-11-14:
The list is only stored on the master. It could be sent to the individual nodes, but they will also do an automatic lookup if using mesh.write().

ricmail85, 2015-11-15:
Thanks for the reply.

How can I build the list from node using write?
Thanks

TMRh20, 2015-11-21:
Sorry, a bit late on reply here. 

To be honest, this is something I've never tried. You would probably want to send the list in a single message, then just store it in an array etc.

ricmail85, 2015-11-22:
Thanks for the reply, I managed following your suggestion. I have another question about the sleepnode function. 
I have a Pir motion detector module and I m able to wake-up the node if I set RISING in the attachinterrupt command in sleepnode. This is because sensor becomes HIGH when truggered. 
I slightky modified your function in order to decide which mode use in attachinterrupt (low, rising...). 
Do you think it can be added to your next release? If you need I can send you my code...

Bye.

TMRh20, 2015-11-23:
@ricmail85 Interesting, I haven't even thought much about using an external source to trigger the interrupt. I think it would probably be easy enough to make the change, but you could also put your code into a pull request. :p

I've been thinking more and more about looking around for a good/well supported sleep and/or low power library like [this](https://github.com/n0m1/Sleep_n0m1) or [this](https://github.com/rocketscream/Low-Power) which may be more beneficial as the Arduino world expands to more and more devices.

ricmail85, 2015-11-23:
I've successfully tested it on my pir detector, I'll tryto do a pull request...
Your sleep function uses the most powerful energy saving settings, so for me it is ok as it is :-)
Bye!

closed 2016-02-21 by Avamander