nRF24/RF24Mesh #45: Code freezes on mesh.update()

franman11, 2015-11-18:
Hello, first things first- I have been using this library recently and while I'm new to the nRF24's themselves, it's been pretty great with the nRF systems I've been building, so thank you!

Now for the cruz of the matter- I have discovered quite a weird problem by accident. I'm using the RF24 Mesh library with a master to switch outputs which receive inputs from sensors attached to 2 remote arduinos, each connected to an nRF24L01+.

The issue is while testing out my setup, I accidentally touched the nRF pin header connected to one of the **sensor nodes** and this caused my **master device** to freeze. After a tormenting time trying to find the cause, I realised the *network.update()* function, found inside the "mesh.update()" function caused an error.

I got this by printing messages on Serial in the following **while** loop:-
                    *while ( radio.isValid() && radio.available(&pipe_num)*
The result- the serial output kept on being printed non-stop. I might add, I printed the outputs of " *radio.isValid()* and  *radio.available(&pipe_num)* and got the following results continuously:-

"radio.isValid() = 1"
"radio.available(&pipe_num) = 1"
"pipe_num = 4"

I believe I can deduce this problem is caused by a corrupt incoming packet (I'm open to correction), but what I fail to understand is why it causes the master device to freeze. It seems to me, there is always something being written on pipe no. 4 (I hope I got the nRF24 basics right) and the code gets stuck in an infinite loop.

Now I'm sure this problem can be eliminated by being careful and keeping my hands off what they shouldn't be on (the nRF pins, in this case) but I'm concerned this issue might pop-up and freeze the automation system that I'm building for my room.

I'm attaching my *.ino* file so you can see the code is quite simple.

Hope you can help me out with this.
Thank you!

[Mesh_Master.txt](https://github.com/TMRh20/RF24Mesh/files/37703/Mesh_Master.txt)

TMRh20, 2015-11-21:
Hi @franman11 thanks for the detailed trouble description and code etc.

I am assuming that you are working from a breadboard, etc. which, I assume, is probably more of a contributing factor to the issue you are describing than anything.

The reason I say this is that the nrf24l01+ modules are mostly self-contained, and manage all aspects of low-layer (physical) communication, including CRC (error) checking.

This being the case, if you are repeatedly receiving corrupt data, most likely due to a connection being interrupted and/or disconnected. For a simple test, I can touch all the pins of my RF24 modules with no problems, while they are communicating. The only impact I see, is a slight delay in communications, since it may interfere slightly, resulting in a higher number of retries or even a few failures, depending on signal strength etc.

TMRh20, 2015-11-21:
Just a thought, since you seem to have been doing a bit of debugging, could you also print out the value of frame_size in the network.update() function when this is happening? I'm curious as to what exactly is happening...

Avamander, 2015-12-14:
Same happened for me when I encountered the fake + original combo. What modules are you using?

Also, do the modules have 100uF electrolytic capacitors soldered to the power pins of the NRF24L01+?

closed 2016-02-21 by Avamander

haseebehsan, 2018-02-15:
Can anyone tell me the solution for that I'm facing the same issue with my module
even the example code isn't working, what should i do? 
P.S: I'm new to it,

ristomatti, 2018-02-15:
@haseebehsan I'm pretty new to this also but based on my recent experiences I've seen the same modules work without any issues on certain days in the same configuration they didn't work some other day. At least triple check the wires if they are on correct pins or have proper contact. I've found myself tediosly checking the wiring several times only to find two hours later I still had two wires reversed. The issue could be similar to the freezing reported on this issue or just being unable to find other nodes on the network.

The other day I noticed if I attach a particular DC power adapter to my breadboard power supply, all modules attached to the same breadboard disconnected. When the power was detached, packets started to fly again.

haseebehsan, 2018-02-17:
@ristomatti thanks, I found the solution earlier. Forgot to update, it was as you said **power supply issue** , power required by nRF24L01 is sometimes not sufficient enough because (in my case) Arduino Nano is powering it self, nRF24L01 module and some sensors. I'm gonna try and connect a **external DC power supply** or I've found a way of connecting a capacitor ranging from **4.7μF to 47μF** between 3.3V and GND pins.