nRF24/RF24Mesh #46: Master node stops working after 12 receives

BennyM, 2015-11-20:
Hi

I've deployed the RF24_MeshExample and RF24_MeshExample_Master on two Arduino Uno's. It works for exactly 12 messages from node 1 to the master node 0. After 12 sends master node 0 stops receiving messages. Only after a power off / on of the master node, messages are sent again.

[image]

BennyM, 2015-11-20:
Forget I said anything

closed 2015-11-20 by BennyM

annawidera, 2016-04-29:
Hi BennyM!
Have you discovered what caused your problem? I've encountered something very similar. I deployed mesh network using Bare Conductive boards with nrf24L01+ connected to them (base of the Bare Conductive is Arduino Leonardo with ATMega32U4). Arduino Uno is programmed as a Master. 
Bare Conductive boards keep sending new values but after a 647 readings Arduino Uno stops receiving any messages (network.available() returns false). 

When I have 2 slaves connected firstly one of them stop being listened by the master (after 50 messages sent) and then (about 600 readings later) the second one also. 

It looks like the connection between the slave and the master is being lost, but the slave doesn't recognize that. Mesh.write() keeps returning true despite the fact that the master doesn't receive any of the messages. 

Thanks, 
Ania

annawidera, 2016-04-30:
Now I think I have nrf24L01+ power lines filtered properly with 2.2uF electrolytic capacitor and 2 ceramic capacitors of value 10nF. 

And still transmission stops after a couple of seconds. Transmitters keep sending data, mesh.write returns true. The slaves are not trying to renew address, because they simply don't know that the connection have been lost.

I think I will create new issue because I am not sure if our problems are related. (Specially because my slave boards don't even try to renew addresses).