nRF24/RF24Mesh #49: setChild() function

ricmail85, 2015-11-29:
Dear TMRh20,

I was looking at your new setChild function. Is it correct? Because in RF24Mesh.cpp, I don't see any difference if the "allow" parameter is true or false... I.e. it seems that the FLAG_NO_POLL is not triggered by the "allow" parameter...

I'm doing it wrong?

TMRh20, 2015-11-29:
Need to ensure rf24network version is 1.0.2 in lib manager

ricmail85, 2015-11-29:
Hi, 
yes I updated both libs... But if you see the setChild func, the input parameter "allow" is never used... Is it right? How the func would know if I want to enable or disable children for this node?

TMRh20, 2015-11-29:
¬_¬  ... Well it worked perfectly well half of the time.

Thanks for pointing this out!

ricmail85, 2015-11-29:
Thank you for your fast fix!

closed 2015-12-14 by Avamander