nRF24/RF24Mesh #50: Multiple nodes (with same ID) in mesh address list

ricmail85, 2015-12-02:
Hi,

I'm finding several issues in the last release. 
In my project, each time I boot a node, it starts with the temporary 255 nodeID and then the master reassign a unique nodeID in the range 1-254. 
Now it happens that if I boot (not at the same time) several nodes, the master results a mesh address list in which the 255 nodeid is listed several times... 
Is it due to the fact that now (in several functions, e.g. setAddress()) the nodeID is declared as a char?

Thanks in advance!

TMRh20, 2015-12-02:
That is the expected behaviour with duplicate nodeids.

Avamander, 2015-12-02:
Why are you dynamically assigning NodeIDs?

ricmail85, 2015-12-03:
Ok, thanks for the reply.

I use dynamic nodeids because the number of nodes is not fixed and I want to avoid any user interaction with the single nodes, all the job is done by the master.

In previous releases, if two nodes had the same nodeid there was an unique id listed on the master... Is it right? Why now they are listed several times?

ricmail85, 2015-12-03:
Ok, I've checked and it is related to func setAddress(char nodeID, uint16_t address). If I change it to setAddress(uint8_t nodeID, uint16_t address), there are no duplicate nodeIDs.
Probably the same applies for the setStaticAddress(char nodeID, uint16_t address)... the nodeID parameter should be uint8_t, not char.

is it right?
bye.

TMRh20, 2015-12-05:
Hmm, looking at this function I'm wondering why its using a char. Will have to get it changed soon.

Avamander, 2015-12-15:
Many thanks to @TMRh20 for fixing this issue and @ricmail85 for reporting it.

closed 2015-12-15 by Avamander

beuleal, 2017-03-22:
I have a similar app. I know that this topic is already closed but when i use setAddress(uint8_t nodeID, uint16_t address), function, it should only change the nodeID based in the address, right? I fell that the function is adding the new module into the module's list rather than edit the nodeID.

Avamander, 2017-03-22:
Well, looking at the examples, I'd say one sets the address before the node connects.

KevinLod, 2018-08-21:
Im working at a project with this awesome library,  but i assigned nodes id manually, i want to automat this by using the DHCP function. But the example codes aren't functioning well, would any body direct me to the right way please (im nooooobe)

beuleal, 2018-08-21:
@KevinLod My solution was create a commom ID, like 1234. When master receive any packet from 1234, it looks his node table i created and assign another ID, like 9234, and the old node 1234 restarts itself with the newer ID 9234.