nRF24/RF24Mesh #51: Maximum number of nodes on network

efsantana, 2015-12-03:
One can only assign nodeIDs from 0 to 255. 

Does this mean that the network can only have 256 nodes? How can I have more nodes and still uniquely identify them?

TMRh20, 2015-12-03:
> Does this mean that the network can only have 256 nodes?

Yup, thats about the max though I doubt the mesh would function well with
that many nodes.

> How can I have more nodes and still uniquely identify them?

Use RF24Network with multicast disabled (780 nodes max)

efsantana, 2015-12-03:
And why is there this limitation?

TMRh20, 2015-12-05:
Limited radio pipes.

1
4
16
64
256
Total: 341

1
5
25
125
625
Total: 781

https://www.nordicsemi.com/chi/content/download/2730/34105/file/nRF24L01_Product_Specification_v2_0.pdf

closed 2015-12-14 by Avamander

robertiko07, 2016-04-09:
nodes on network is about 781, but nodes IDs?