nRF24/RF24Mesh #53: Continious send to test antenna.

IgorTheWeak, 2015-12-16:
Hi!

I was wondering what would be the easiest way to put radio in continious transmitt mode, so I can test my antenna on different channels.

I tried just looping mesh.begin and mesh.update but it is not true continious transmit and it takes time for analyzer to show result.

I know that nordic datasheet gives example of this, but it is bare C and assembler code.

I believe this is very easy.
Thanks in advance!

Avamander, 2015-12-16:
Don't use RF24Mesh, use RF24 for that.

IgorTheWeak, 2015-12-17:
This is what Nordic does when they need continouous transmit carrier.

Set PWR_UP=1
Delay 1.5ms (wait to power on).
Set the chip to be Primary-TX
Turn off all auto acknowledgement functionality
Set the RF channel you want
Set your output power
Set PLL_LOCK to 1
Configure TX address as 5 bytes with all 0xFF
Fill the TX payload with 32 bytes of 0xFF
Turn off CRC
Send the packet by pulsing CE
Wait until the transmission ends
Set CE high
Use the SPI command for re-use of last sent packet (0xE3)
Keep CE high as long as you want the carrier

I am just not that familiar with this rf24 libraries content to execute this code. 

I tried just putting   radio.write( &carrier, sizeof(int) ); in loop, but didnt work that well.

TMRh20, 2015-12-17:
The RF24 lib doesn't include a function to set PLL lock to 1.

Beyond that, you might be able to fake it or do a standard transmit by using startFastWrite() and call [reUseTX()](http://tmrh20.github.io/RF24/classRF24.html#aeaf7fa54d3ab2a85ce215b4bf6ae933b) function.

Avamander, 2016-01-13:
@TMRh20 Is the PLL lock worth implementing? Does it give a performance or other bonus to RF24?

TMRh20, 2016-01-13:
As far as I know its just for testing, to have the module output a steady carrier signal. Implementing a function to enable it shouldn't interfere with the overall API, or add any additional prog space etc. unless used, so I don't see why not. Some may find it useful.

IgorTheWeak, 2016-02-01:
I simply needed to test my pcb antenna with casing. And if the carrier is steady and constant it is easy to tune the antenna for best output power for given frequency. Otherwise the spectrum analyzer gives readings rather slowly.

closed 2016-02-21 by Avamander