nRF24/RF24Mesh #55: Mesh on RPi not working

asollander, 2015-12-28:
I'm developing an application with a Raspberry Pi 2 as a master, and several Arduino Nano as nodes.

If I try using the basic mesh examples with an Arduino as the master, it works fine. If I use the RPi as a master, the nodes keep trying to renew their addresses, but nothing happens.

If I then start the Arduino master alongside my RPi (yes, I know there should only be one master), I start getting the messages to the RPi too.

Working with the RF24Network examples work like a breeze, so I assume it's not a hardware problem.

Regards,
Anders

TMRh20, 2015-12-28:
If everything else is working, the first thing that comes to mind is to delete the dhcplist.txt file found in the sketch directory. Mesh address assignments are saved on RPi, so if an old list is present, it may block new/additional nodes from joining.

Beyond that, unplug/powercycle the radio module to ensure it powers up with fresh settings. You can enable debugging at the RF24Mesh and/or RF24Network layers on the node and master to see if there is any communication going on also. If no luck please post the startup and debug output of the master node.

asollander, 2015-12-28:
How do I get this startup information? I only know of the printDetails (though I've turned on the debugging macros). The only messages I get are occasional

MSH: Invalid id 0 rcvd.

This is the output of details.

================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
TX_ADDR          = 0xcccccc3ecc
RX_PW_P0-6       = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA            = 0x3e
EN_RXADDR        = 0x3f
RF_CH            = 0x61
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX

TMRh20, 2015-12-28:
You got it, that is the startup info I was looking for. There should be a bunch more info coming from RF24Network. With a single node using the default example, it should look like this:

```
...
CRC Length       = 16 bits
PA Power         = PA_MAX
4293970880: MAC Received on 0 id 5991 from 04444 to 0100 type 194
4293970881: MAC Sending to 04444 via 04444 on pipe 0
4293970882: NET Pipe 0 on node 04444 has address cccccc3ecc
4293970884: MAC Sent on cccc3ecc ok
4293970936: MAC Received on 0 id 5991 from 04444 to 00 type 195
4293970939: NET Sending id 5991 from 00 to 04444 type 128
4293970940: FRG frame size 10
4293970941: FRG frame 00 00 24 09 67 17 80 01 04 00
4293970942: MAC Sending to 04444 via 04444 on pipe 0
4293970943: NET Pipe 0 on node 04444 has address cccccc3ecc
4293970945: MAC Sent on cccc3ecc ok
4293970962: MAC Received on 4 id 5991 from 04 to 00 type 129
Sent to 04444 phys: 04444 new: 04 id: 1
```

Type 194 = NETWORK POLL
Type 195 = ADDRESS REQUEST
Type 129 = ADDRESS CONFIRMATION
Assign address 04 to nodeID 1:  ``` Sent to 04444 phys: 04444 new: 04 id: 1 ```

Make sure to run ``` sudo make install -B ``` after enabling debug in the RF24Network_config.h file, to ensure it fully re-compiles also.

asollander, 2015-12-28:
I realize I only turned on the debugging for RF24Mesh, and not for RF24Network. I still haven't turned it on for RF24, but I assume that's not necessary. Here's the output from the master (on RPi) when I;ve turned on debugging, and an Arduino client (node ID 16) is trying to connect (sorry for the length -- was unclear where to cut):

setup_address node=00 mask=00 parent=00 pipe=00
112: NET Pipe 5 on node 00 has address cccccccce3
112: NET Pipe 4 on node 00 has address cccccccc3e
112: NET Pipe 3 on node 00 has address ccccccccce
112: NET Pipe 2 on node 00 has address cccccccc33
112: NET Pipe 1 on node 00 has address cccccccc3c
112: NET Pipe 0 on node 00 has address ccccccccc3
================ SPI Configuration ================
CSN Pin          = CE0 (PI Hardware Driven)
CE Pin           = Custom GPIO22
Clock Speed      = 8 Mhz
================ NRF Configuration ================
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5     = 0x33 0xce 0x3e 0xe3
TX_ADDR          = 0xcccccc3ecc
RX_PW_P0-6       = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA            = 0x3e
EN_RXADDR        = 0x3f
RF_CH            = 0x61
RF_SETUP         = 0x07
CONFIG           = 0x0f
DYNPD/FEATURE    = 0x3f 0x04
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
314: MAC Received on 0 id 25 from 04444 to 0100 type 194
314: MAC Sending to 04444 via 04444 on pipe 0
314: NET Pipe 0 on node 04444 has address cccccc3ecc
315: MAC Sent on cccc3ecc ok
515: MAC Received on 0 id 27 from 04444 to 00 type 195
515 MAC: System payload rcvd 195
716: MAC Received on 0 id 38192 from 04444 to 00 type 195
716 MAC: System payload rcvd 195
916: MAC Received on 0 id 38193 from 04444 to 0100 type 194
917: MAC Sending to 04444 via 04444 on pipe 0
917: NET Pipe 0 on node 04444 has address cccccc3ecc
918: MAC Sent on cccc3ecc ok
918: MAC Received on 0 id 29 from 04444 to 0100 type 194
919: MAC Sending to 04444 via 04444 on pipe 0
919: NET Pipe 0 on node 04444 has address cccccc3ecc
920: MAC Sent on cccc3ecc ok
920: MAC Received on 0 id 32 from 04444 to 0100 type 194
920: MAC Sending to 04444 via 04444 on pipe 0
921: NET Pipe 0 on node 04444 has address cccccc3ecc
921: MAC Sent on cccc3ecc ok
1122: MAC Received on 0 id 35 from 04444 to 00 type 195
1122 MAC: System payload rcvd 195
1323: MAC Received on 0 id 36 from 04444 to 0100 type 194
1323: MAC Sending to 04444 via 04444 on pipe 0
1323: NET Pipe 0 on node 04444 has address cccccc3ecc
1324: MAC Sent on cccc3ecc ok
1524: MAC Received on 0 id 38201 from 04444 to 0100 type 194
1525: MAC Sending to 04444 via 04444 on pipe 0
1525: NET Pipe 0 on node 04444 has address cccccc3ecc
1526: MAC Sent on cccc3ecc ok
1526: MAC Received on 0 id 38201 from 04444 to 00 type 195
1527 MAC: System payload rcvd 195
1727: MAC Received on 0 id 38204 from 04444 to 0100 type 194
1727: MAC Sending to 04444 via 04444 on pipe 0
1728: NET Pipe 0 on node 04444 has address cccccc3ecc
1729: MAC Sent on cccc3ecc ok
1729: MAC Received on 0 id 38204 from 04444 to 00 type 195
1729 MAC: System payload rcvd 195
1930: MAC Received on 0 id 40 from 04444 to 0100 type 194
1930: MAC Sending to 04444 via 04444 on pipe 0
1930: NET Pipe 0 on node 04444 has address cccccc3ecc
[... 84 lines cut ...]
4669: NET Pipe 5 on node 00 has address cccccccce3
4702: MAC Sent on cccccce3 failed
4702: MAC Send fail to 00 via 00 on pipe 5
********Assigned Addresses********
**********************************
5149: MAC Received on 0 id 38239 from 04444 to 0100 type 194
5149: MAC Sending to 04444 via 04444 on pipe 0
5150: NET Pipe 0 on node 04444 has address cccccc3ecc
5150: MAC Sent on cccc3ecc ok
5151: MAC Received on 0 id 75 from 04444 to 0100 type 194
5151: MAC Sending to 04444 via 04444 on pipe 0
5151: NET Pipe 0 on node 04444 has address cccccc3ecc
5152: MAC Sent on cccc3ecc ok
5352: MAC Received on 0 id 38241 from 04444 to 00 type 195
5352 MAC: System payload rcvd 195
5553: MAC Received on 0 id 77 from 04444 to 00 type 195
5553 MAC: System payload rcvd 195
5753: MAC Received on 0 id 79 from 04444 to 0100 type 194
5753: MAC Sending to 04444 via 04444 on pipe 0
5754: NET Pipe 0 on node 04444 has address cccccc3ecc
5754: MAC Sent on cccc3ecc ok
5755: MAC Received on 0 id 38243 from 04444 to 0100 type 194
5755: MAC Sending to 04444 via 04444 on pipe 0
5755: NET Pipe 0 on node 04444 has address cccccc3ecc
5756: MAC Sent on cccc3ecc ok

TMRh20, 2015-12-29:
The log indicates that at least 2 devices are trying to connect, but the master is not responding with any available addresses. The master is also sending address responses to itself, suggesting dual master nodes are running (bad things).

It also suggests that communication is working fine, but the master is refusing to assign addresses, which should be resolved by disconnecting power from every radio, deleting the dhcplist.txt file, and connecting only one node and one master.

asollander, 2016-01-04:
I've continued debugging these issues, but haven't found a solution yet. I did notice, however, that whereas for my RF24Network tests, I get

  TX_ADDR          = 0xe7e7e7e7e7

and for RF24Mesh

 TX_ADDR          = 0xcccccc3ecc

I realize that the channel is different (as long as I keep using the defaults, but I even tried changing the channel for the mesh to be the "normal" standard), but this doesn't seem to make a difference.

Could this be connected to the malfunctioning here?

Thanks,
Anders

TMRh20, 2016-01-05:
Unfortunately not, it will show like that at first, but if calling printDetails() after some activity, the RF24Network tx_addr will look similar.

RF24Network/Mesh use two main features, dynamic payloads, and pipe-specific auto-ack settings. To test functionality at a basic level, I would suggest running the core RF24 examples to test specific functionality between RPi and Arduino. Both gettingstarted sketches and the pingpair_dyn examples will give you a good idea if things are working or not. 

If it all checks out, try the main gettingstarted sketch with these lines added on both RPi and Arduino:
```
radio.setAutoAck(0,0);
radio.setAutoAck(1,0);
```

If you take a look at [this comment](https://github.com/TMRh20/RF24/issues/182#issuecomment-167337905) and the above post with pictures, there is some good info on modules. I would start to assume that there is something with your modules if the above is problematic..

asollander, 2016-01-06:
I tried what you suggested, and working with only Arduinos I had no problems with gettingstarted. RPi/Arduino combination did prove complicated, though. gettingstarted didn't work, but pingpair_dyn worked perfectly.

After turning off auto acks for the pipes as you suggested, gettingstarted worked perfectly with Arduino/RPi -- in both directions.

Do you think it could work to just turn off Acks altogether for the Mesh functionality?

TMRh20, 2016-01-06:
Try gettingstarted again, but make sure to disconnect the power to the modules first. This makes sure they are powered up with default settings.

If you still have problems after a fresh power-up then it would seem like the hardware does not support auto-ack. A network config without auto-ack probably wouldn't work all that well.

I would also suggest updating your libraries on both the Arduino and RPi to the latest versions of RF24Network and RF24Mesh, since some past updates caused compatibility issues between versions.

asollander, 2016-01-06:
Ok, I'll try this.
I actually just tried using 
```C++
radio.setAutoAck(false)
```
on both the RPi and Arduino, and they just crashed there (SIGSEGV). 

I'll try your suggestions with the fresh power-up and updates libraries. Thanks!

asollander, 2016-01-08:
So, I've now updated the libraries to be the latest and greatest, but the problems prevail.

What could be the reason that I can send data back and forth, but there is no HW acknowledgement? You suggested this may be due to my hardware not supporting auto-ack. Do you mean my RF24 chips or Arduino/RPi in this case?

TMRh20, 2016-01-08:
The results seem too inconsistent to say.

Pingpair_dyn uses auto-ack, so there is no reason it should work with that sketch and not the gettingstarted sketch.

I would suggest attempting again with batteries, or using the following method from [here](http://nerdralph.blogspot.ca/2014/01/nrf24l01-control-with-3-attiny85-pins.html)

> For powering the nrf24l01+ module, a 3.3v regulator could be used, but a cheaper and simpler way is to drop a 5V supply to 3V using a 20mA-rated red led.  Most red LEDs have a forward voltage drop between 1.8 and 2.2V, leaving 3.2-2.8V for the nrf, which is well within the 1.9-3.6V requirement.

asollander, 2016-01-12:
Turned out it was the libraries after all. A clandestinely installed Arduino IDE pointed to libraries where I didn't have them, i.e. not the newest. It's now resolved and runs much more smoothly, though there are still glitches. Thanks for all your help finding this out.

closed 2016-01-12 by asollander

patchido, 2016-04-14:
hi, i am trying to use this library which in fact is awsome, great work. The application im trying to achieve is to have 10 nodes on arduino unos reporting to a Rpi, i just need to listen to the nodes, not talk to them. so i had success in the example code, but for some reason NodeID =1 appears as 5 on the RPI, this is really important for me to be correct, and then NodeID =2 appears like 4, and so on. until i get to node 5, node 6 wont work after that(trying one at the time) is this a limitation? can't i have 10 nodes reporting in?

Avamander, 2016-04-14:
Are you maybe confusing RF24Network addresses with NodeIDs? 

The mesh should support hundreds of nodes, the reason it doesn't work is the code you're using. Please try loading the examples onto all your devices and see if it works.