nRF24/RF24Mesh #58: "Dead nodes" linger on in RF24Mesh DHCP table

asollander, 2016-01-12:
When I have nodes that leave the network (actively powered off), they still seem to linger on in the network, and no matter how long I wait, the mesh doesn't seem to look for stale entries. Is there a way to enforce this?

Thanks,
Anders

Avamander, 2016-01-13:
Mesh doesn't do that automatically. You have to manually release the
address on the node that goes offline.

asollander, 2016-01-13:
I saw the method releaseAddress, but is there a corresponding method for the master, in case where the nodes get unresponsive?

Avamander, 2016-01-13:
Null the address in the node list. But you really should make sure nodes
release their addresses.

TMRh20, 2016-01-13:
The [setAddress()](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a6aafd9dbdd198487de392f9992218f04) function can be used to manually manage addresses in cases where the master is able to 'detect' inactive nodes via inactivity etc.

leviso1981, 2016-01-24:
Hi TMRh20. How to use setAddress() to clean a nodeID from the mesh AddrList?

closed 2016-02-07 by Avamander