nRF24/RF24Mesh #60: The command network.write does not work properly in mesh network

leviso1981, 2016-01-25:
Hi

I have a mesh network with 5 devices: 1 master (raspberry Pi) and 4 sensors (Arduino UNO R3). The master sends a hello packet every 10 seconds to all sensors in order to monitor whether all are responding properly.
But when a sensor gets a "child" mesh address like "044", apparently the sensor with Mesh address "04" is answering the hello packet directly.
To clarify, follows the configuration of the network:
NodeID: 11 - MeshID: 044
NodeID: 22 - MeshID: 03
NodeID: 33 - MeshID: 04
NodeID: 44 - MeshID: 02

When the master sends out a hello packet for the sensor with MeshID: "044", the packet is sent successfully even if this sensor is off! The sensor with MeshID "04" is responding directly without consulting the sensor with MeshID "044".

I believe that the network should not work on this way, correct? Is there any way to fix this issue?

I am using "network.write" to send the hello packet from master to sensors.

tjb8907, 2016-01-27:
I would try using mesh.write instead.

leviso1981, 2016-02-02:
Hi Tjb8907,

I think that "mesh.white" is used for communication from nodes to the master only. isn't It? For communication from Master to nodes, we need to use the network.write command.

The problem continues. Has anyone experienced this before?

Thanks for any help.

Interneedus, 2016-02-02:
https://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#ad294234c41f303881aa24c74782eecd7

You can send messages to other nodes too.

>

leviso1981, 2016-02-03:
Hi,

I was using an older version that doesn't support the command "mesh.write" when sending packets from master to sensor nodes.

With the update, really works sending my hello packet from the master to nodes with mesh.write. Problem solved!

Thanks for spending time to help me!

closed 2016-02-03 by Avamander