nRF24/RF24Mesh #63: Nodes not reconnecting

sybrenromer, 2016-02-09:
Hi,

I was very happy when I found this library, as it offers exactly what I need for my sensor network!
I still have one problem that I haven't been able to figure out.

When connecting my first arduino to my Raspberry Pi, it is assignmed address 04. When connecting a second node, it gets 044. If I remove the first node, the second is not able to reconnect.
Clearing dhcplist.txt doesn't help.

Has onyone an idea about what might be going on?

I haven't tested it with a third node yet, but I will do this and post the results here if I haven't found the answer.

I'm happy to answer any questions!

Avamander, 2016-02-09:
Are you sure it has enough transmission power to reach the Pi?

sybrenromer, 2016-02-09:
Good question. In the previous setup I had both arduino's next to each other.
I also read that the mega has a poor 3.3 volt regulator, so this could have been the problem.
Now, however, I put the raspberry pi in the middle and both nodes in a different direction a few meters away.
It still behaves in the same way.

When reading the serial output, it does say that it wants to reconnect, but this doesn't work.
As soon as the first node comes online again, the second starts transmitting as well.

Avamander, 2016-02-09:
Have you properly set-up NodeID's (starting from one, not 0 and
incrementing) and what code are you running on the second node (the one
that needs the first node)?

sybrenromer, 2016-02-09:
I used the exact same code, with the only change being the nodeID and the CE/CS pins.

I'm starting the get the idea that the second arduino has problems, but that doesn't make sense as it can send fine when the first one is also connected.

When I run the RF24Mesh example on the arduino and the Example Master on the Pi, it still doesn't work.

I will try it with a different arduino to see if that helps.

sybrenromer, 2016-02-09:
Using a different arduino doesn't make a difference.
When assigning a different NodeID to it, the address changes.
After doing this 4 times, the addresses 04, 044, 034, 024, 014 are used and the last time it can't get an address!

The terminal fills up with messages like this:

ID: 1 ADDR: 04
ID: 2 ADDR: 044
not allocated
ID: 1 ADDR: 04
ID: 2 ADDR: 044
ID: 3 ADDR: 034
not allocated
ID: 1 ADDR: 04
ID: 2 ADDR: 044
ID: 3 ADDR: 034
ID: 4 ADDR: 024
not allocated
ID: 1 ADDR: 04
ID: 2 ADDR: 044
ID: 3 ADDR: 034
ID: 4 ADDR: 024
ID: 5 ADDR: 014
not allocated
ID: 1 ADDR: 04
ID: 2 ADDR: 044
not allocated

It's almost as if the master node only accepts one child, even though this should be 4.
I'm really curious what happend when adding a third or even fourth arduino, but I have to wait for more components first.

sybrenromer, 2016-02-09:
I've done some more testing and I suspect the problem is really in the Arduino Megas.

When I change the nodeID of the nano, it all works fine, even with re-addressing and everything.
Unfortunately I still have to wait a couple of days before I receive any extra nanos.

If I find out anything, I will post it here.

Interneedus, 2016-02-09:
Usually the problems with Mega can be solved by soldering a 100uF (until
2500uF) electrolytic capacitor to the VCC and GND pins of the NRF.

I'm not exactly understanding, but I'll repeat just in case. Do not(!)
manually assign RF24Network addresses, only change nodeid in the range from
1 to 255. NodeID is like a MAC address, RF24Network address is like IP
address. (Kinda)

Second thing is that I'd try wiping the entire install, redownload every
core library, why not even the IDE too. Do the same on the RPi. There might
be some weird leftover file causing this.

closed 2016-03-23 by TMRh20