nRF24/RF24Mesh #64: Two networks nearby

Fctoni, 2016-02-15:
Hello!

I was wondering if two meshes can work nearby, for example have two masters and several nodes that can all see each other. Is there a way to set a unique address or ID to each network and make each node only "see" the nodes of same network and prevent one network from seeing the other?

Thanks!

Avamander, 2016-02-15:
Using different channels it should be possible.

Fctoni, 2016-02-16:
Thanks for prompt reply. Anyone has tested this yet? How many channels are available? Thank you!

TMRh20, 2016-02-20:
It is a feature of the [mesh.begin()](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a4159edcb2a13c713cd9c524315270902) function.

closed 2016-02-21 by Avamander

fireghostea, 2017-10-27:
hello
how should i assign channel dynamically
can i check for existing near mesh and base on the result define channel for not conflicting ?

Avamander, 2017-10-28:
> can i check for existing near mesh and base on the result define channel for not conflicting ?
Yes.

fireghostea, 2017-10-29:
ok how should i do it?

MrBeezel, 2018-04-16:
Hello,
I'm new to this library and I want to ask about a particular situation... If we have two independent mesh networks whose masters are not close, but at least one of the secondary nodes of each network does it, could it cause some kind of of problem?

Obviously, both networks are supposed to work on the same channel for my example.

My idea in this situation is to (somehow) detect that there are a second active mesh and then change the channel to avoid the interference but i dont know if is possible because maybe the mesh will be affected before i can detect the new network (for example, by changing nodes from one mesh to another one or finding repeated IDs)

What do you think about this?

(Also, thanks for this library, is amazing!)