nRF24/RF24Mesh #66: RPi 2 Mesh Master No Debug Msg

jetiston, 2016-02-24:
Hi. I have a Raspberry Pi 2 as master and an Arduino as my node. I have tried examples from RF24 library (gettingstarted_call_response) and RF24Network library (helloworld_tx/rx) and they work fine. 

I enabled the MESH_DEBUG in mesh config and SERIAL_DEBUG and SERIAL_DEBUG_ROUTING in network config but nothing is output in the RPi when I run the RF24Mesh_Example_Master except the start up details
`sudo ./RF24Mesh_Example_Master 
start
setup_address node=00 mask=00 parent=00 pipe=00
2080213: NET Pipe 5 on node 00 has address cccccccce3
2080213: NET Pipe 4 on node 00 has address cccccccc3e
2080213: NET Pipe 3 on node 00 has address ccccccccce
2080213: NET Pipe 2 on node 00 has address cccccccc33
2080213: NET Pipe 1 on node 00 has address cccccccc3c
2080214: NET Pipe 0 on node 00 has address ccccccccc3
================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO22
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xcccccccc3c
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x61
RF_SETUP	 = 0x07
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX`

In the Arduino side it outputs a sort of network poll but it just do it over and over again. 

What does that mean? or what could I be doing wrong?
Thanks in advance.

TMRh20, 2016-02-24:
You should be seeing something like this:

```
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
1739178: MAC Received on 0 id 26614 from 04444 to 0100 type 194
1739179: MAC Sending to 04444 via 04444 on pipe 0
1739179: NET Pipe 0 on node 04444 has address cccccc3ecc
1739180: MAC Sent on cccc3ecc ok
1739235: MAC Received on 0 id 26614 from 04444 to 00 type 195
```

Make sure to do a full recompile after modifying the debug options: make install *-B* or make clean

If you still see nothing, even with RF24Network debug enabled, then the RPi is just not receiving anything... printDetails() output looks good...

jetiston, 2016-03-04:
Nothings happens. I wonder if it the arduino is the one not sending anything. But it outputs this when I run the RF24Mesh_Example 

48 MSH: Poll 
49: NET Sending id 1 from 04444 to 0100 type 194
53: MAC Sending to 00 via 00 on pipe 0
56: NET Pipe 0 on node 00 has address ccccccccc3
61: MAC Sent on ccccccc3 ok
119 MSH: No poll from level 0

so it's safe to assume it's sending something right?

But I have a question regarding that output. It is sending a packet from 04444 (w/c is the node) to 0100 but it sends it to 00 which is the RPi. So maybe the RPi doesn't recognise it as a packet for itself? or is my interpretation wrong?

TMRh20, 2016-03-23:
0100 is just a false address used to indicate a multicast payload
04444 is the default address for renewal

It looks ok, just not getting a poll response

jetiston, 2016-03-29:
Okay. So it seems that it actually doesn't receive anything.
I tried doing network.multicast. When the rpi master is doing the multicast the arduino nodes receives but when the node does a multicast nobody receives it. I think that's why there's no poll response.

Why do you think this happens?

TMRh20, 2016-03-29:
I'm guessing one of your devices just isn't working properly. I would suggest testing down the stack a bit, I usually like to start at the bottom and work my way up when things are just not working.

1. Load up the basic RF24 gettingstarted_call_response sketch
   a: Ensure the RF24 radio(7,8); is using the correct pins
   b: If using RPi and Arduino, the radio numbers should be pre configured.
   c: Either power cycle the Arduinos after uploading, or Add the line radio.setAutoAck(1); after radio.begin() to ensure AA is enabled after using RF24Mesh

If that works, you know your radios are connected properly and functioning. If not, most likely a config/hookup/hardware issue.

If not working, we need to get the printDetails() output, so:
a: Add this to the top includes to get basic debug info on Arduino:
```
#include <SPI.h>
#include "RF24.h"
#include <printf.h>
```
in setup add:
```
printf_begin();
```
anywhere after Serial.begin();

and then uncomment ```radio.printDetails() ``` to get debug output on Arduino.

else:

2. Test with RF24Network helloworld examples similarly if working at RF24 layer. Make sure the pins in RF24 radio(7,8) are set correctly, everything else is configured.

3. If the above all works, RF24Mesh should also work...

Please keep adding to this issue, even if issues are found at lower layers. If resolved, it may help others with similar issues.

closed 2019-05-10 by TMRh20