nRF24/RF24Mesh #70: Missing license statement

bluelightning, 2016-03-20:
RF24Mesh doesn't have anything stating what license it can be distributed under. I presume GPLv2 since it appears to be sort of derived from RF24Network, but it would be preferable to have a clear statement - e.g. comments at the top of each source file / header.

TMRh20, 2016-03-29:
Yes, licensing needs review. 
Please see https://github.com/TMRh20/RF24/issues/227 and https://github.com/TMRh20/RF24Ethernet/issues/13

Avamander, 2016-06-17:
@TMRh20

Also concerns https://github.com/TMRh20/RF24/issues/227 and https://github.com/TMRh20/RF24Ethernet/issues/13

I looked into the thing. 

> When we say that GPLv2 and GPLv3 are incompatible, it means there is no legal way to combine code under GPLv2 with code under GPLv3 in a single program. This is because both GPLv2 and GPLv3 are copyleft licenses: each of them says, “If you include code under this license in a larger program, the larger program must be under this license too.” There is no way to make them compatible. We could add a GPLv2-compatibility clause to GPLv3, but it wouldn't do the job, because GPLv2 would need a similar clause.

This means we can license the full stack (RF24*) as GPLv2. Unless maniacbug comes and jumps in a relicences RF24_config.h, RF24.h, RF24_arch_config.h and a few other sketches. “If you include code under this license in a larger program, the larger program must be under this license too.” 

So should we license everything under GPLv2 or look for ways to get it relicensed?

TMRh20, 2016-06-18:
I haven't had a chance to look into this properly, but there are always a few loopholes.

RF24 and RF24Network are derived from ManiacBugs GPLv2.0 code
RF24Mesh is my design
RF24Ethernet is partly derived from SerialIP and UIPEthernet which are compatible with GPLv3 but not both with v2 I believe

So I'm sure any GPL3 software etc. can *use* RF24 and RF24Network as long as it is not included in the program. Thanks to all the work that has gone into supporting this stuff, it is pretty painless to install via Arduino or RPi/Linux, so I don't think it is too much of an issue.

As always, I could be horribly wrong, but either way, I'll hopefully get a chance to get back to a bit more RF24 maintenance etc. and look at this stuff a bit closer. The issues are piling up :fearful:

Avamander, 2016-06-18:
But when compiled on Arduino IDE and uploaded, is that not using it in "larger program"?

Avamander, 2016-06-18:
Nevermind the previous comment. 

If the copyright holder has disappeared, what happens to his licences stuff? Could we relicence it?

Avamander, 2016-06-18:
Could we ask about this in StackExchange? Maybe more knowledgeable can help?

TMRh20, 2016-06-18:
Sure, the more info the better

Avamander, 2017-04-15:
@TMRh20 @spaniakos @ciberkids @yozik04 @ivankravets @julianpas @l3v5y 

Please comment and say if you'd agree with RF24Mesh being licensed GPLv2+

ivankravets, 2017-04-15:
AGREE with GPLv2+

julianpas, 2017-04-15:
My involvement here is so little that I will not confront whatever licensing decision is made. Also since I have never released anything using this lib as it only went into personal for fun tasks I have no stake in the choice. 

However just from the perspective of my experience in other open source projects I will be really happy if there was a way to have this lib under a more permissive license like MIT or some such. 

Good luck! :)

yozik04, 2017-04-15:
Agree with any license. I think that my involvement was too little to make a choice.

TMRh20, 2017-04-15:
Yeah, probably the least restrictive the better.

Avamander, 2017-04-15:
It doesn't really help a lot if this is MIT and not GPLv2 as the base code is all GPLv2 (by maniacbug).

spaniakos, 2018-05-10:
sorry for the delayed post.
since i am not a main contributor of the library.
I say you can change the license to what you think is appropriate!

closed 2019-05-10 by TMRh20