nRF24/RF24Mesh #71: ESP8266 wdt reset.

gamecompiler, 2016-04-09:
Running the RF24Mesh_Example_Master_Statistics and RF24Mesh_Example sketches on two ESPs now. 
But the Master crashes horrible when receiving data.
I use pin 4 as CE and pin 15 as CSN.
Flashed using Arduino IDE 1.6.6 and ESP8266 "Add-On" 2.0.0.
Id have the RF24
Id not encounter such kind of problems without Mesh functionality, with the pingpair_dyn sketch.

''''
>>>stack>>>
3ffefb50:  000000ff 0000000c 3ffeeaa8 3ffeebe4  
3ffefb60:  000000ff 3ffee92a 3ffeeaa8 00000000  
3ffefb70:  000000ff 3ffee92a 3ffee920 40202190  
3ffefb80:  cccccc04 00000000 3ffee920 40201ed4  
3ffefb90:  3ffeeaa8 3ffeeaa8 3ffeeac8 3ffeebe4  
3ffefba0:  3ffee8fc 00000000 3ffee8fc 40203006  
3ffefbb0:  40100e79 3fff0c88 3fff0c88 3ffeebe4  
3ffefbc0:  3ffee8fc 00000000 3ffeebdc 402018ed  
3ffefbd0:  00000000 00000000 00000016 40101431  
3ffefbe0:  40203f79 3ffeebb8 3ffee8fc 3ffeebe4  
3ffefbf0:  3fffdc20 00000000 3ffeebdc 40203fa1  
3ffefc00:  00000000 00000000 3ffeebf0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
0

Exception (9):
epc1=0x40201bee epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffee969 depc=0x00000000

ctx: cont 
sp: 3ffef9b0 end: 3ffefc10 offset: 01a0
''''

Did you also run into these problems using the esp8266 ?

With king regards.
gamecompiler

TMRh20, 2016-04-10:
Can you check you library.properties file
(documents/arduino/libraries/RF24network) for rf24network and mesh? Should
be v1.0.5 for both else this is a known issue in prev versions.

gamecompiler, 2016-04-10:
Network:1.0.5
Mesh:1.0.5
Ethernet:1.6.1
Rf24:1.6.1 

Id updated all through the Library Manager before submitting this issue.

Could the Rf24 Version be a problem ?

TMRh20, 2016-04-11:
I'm not sure, if you mean RF24 v1.1.6,  I'm not showing any problems with it & similar, but:
IDE: 1.6.7
ESP8266: [Git Version](https://github.com/esp8266/Arduino#using-git-version)

Maybe try updating ESP stuff?

gamecompiler, 2016-04-11:
Ok got it work. Thank you very much for your help :dancer: 
Now im using IDE: 1.6.7, the ESP8266 Add-On git version and the current fetch of your RF24 librarys.

Thank you very very much ! :smile:

closed 2016-04-11 by gamecompiler