nRF24/RF24Mesh #72: Mesh not working

aaddame, 2016-04-12:
I tried to use the mesh code, but it doesn't work
I have a yun working as master and a devduino as node.
The yun runs with  the RF24Mesh_Example_Master code
The devduino runs with the RF24Mesh_Example.

The yun keeps displaying an empty address list.

Using the latest libraries.

aaddame, 2016-04-12:
Went back to version 1.0.3 of both Network and Mesh libraries. 
Now it sort of works:

********Assigned Addresses********
NodeID: 210 RF24Network Address: 0151322
NodeID: 210 RF24Network Address: 0151322
NodeID: 210 RF24Network Address: 0151322
NodeID: 210 RF24Network Address: 0151322
NodeID: 210 RF24Network Address: 0151322
**********************************

THe nodeid of the devduino is defined as 1

aaddame, 2016-04-12:
Version 1.0.3 of both libraries also doesn't work

TMRh20, 2016-04-12:
Can you/have you tested with the RF24Network or RF24 examples?

aaddame, 2016-04-12:
Yes, both work. It did thsi to verify the radios were working

TMRh20, 2016-04-12:
I'm thinking it might have something to do with the Yun being detected as a Linux device. 

In RF24Mesh_config.h can you uncomment ``` #define MESH_DEBUG_MINIMAL ``` and ```#define MESH_DEBUG ``` and post the output of the Yun?

aaddame, 2016-04-12:
In the 1.0.5 version?

TMRh20, 2016-04-12:
Yeah, with the latest updates preferably.

aaddame, 2016-04-13:
Solved.
It was an issue on my build system. I reinstalled arduino ide and libraries and now everything works as it should

 
********Assigned Addresses********
NodeID: 1 RF24Network Address: 045
NodeID: 2 RF24Network Address: 04
NodeID: 3 RF24Network Address: 044
**********************************
216072
406032
139476
217088
407032

closed 2016-04-13 by aaddame

Sharathmk99, 2016-10-08:
@aaddame I'm also trying with YUN.
Can you share the Pin connection for NRF and YUN.