nRF24/RF24Mesh #73: RPI miss indexing nodes.

patchido, 2016-04-14:
hi, i am trying to use this library which in fact is awsome, great work. The application im trying to achieve is to have 10 nodes on arduino unos reporting to a Rpi, i just need to listen to the nodes, not talk to them. so i had success in the example code, but for some reason NodeID =1 appears as 5 on the RPI, this is really important for me to be correct, and then NodeID =2 appears like 4, and so on. until i get to node 5, node 6 wont work after that(trying one at the time) is this a limitation? can't i have 10 nodes reporting in?

im using the master example on RPI the arduino code is changed but if i use arduino as master i get the node correct.

Avamander, 2016-04-14:
Are you sure you aren't mistaking RF24Network's network address for RF24Mesh's NodeID? I'd also again make sure the examples work properly and then step by step try adding more nodes and maybe your code.

patchido, 2016-04-15:
Im using the example code on the RPI as master.

```
while(network.available()){
//    printf("rcv\n");
    RF24NetworkHeader header;
    network.peek(header);

    uint32_t dat=0;
    switch(header.type){
      // Display the incoming millis() values from the sensor nodes
      case 'M': network.read(header,&dat,sizeof(dat));
                printf("Rcv %u from 0%o\n",dat,header.from_node);
                 break;
      default:  network.read(header,0,0);
                printf("Rcv bad type %d from 0%o\n",header.type,header.from_node);
                break;
    }
  }
delay(2);
  }
return 0;
}
```

does header.from_node print the actual nodeID? also, why can i only use nodeId 1-5, 6 wouldn't work, i tried the same actual arduino just changing the nodeId for each, to test.. could that be it? did i finish the 4 nodes that the rpi could hear? if i had more nodes i could actually achieve it?

thanks for your fast response.

TMRh20, 2016-04-15:
Take a look at http://tmrh20.github.io/RF24Mesh/General-Usage.html and http://tmrh20.github.io/RF24Network/Addressing.html 

Header is the network header, containing network addresses. Mesh uses nodeIDs, nodeIDs 1-255 are supported

closed 2016-04-17 by Avamander

DUILIOJR, 2016-09-20:
Hi all, I got excatly the same behavior. Using the example code on the RPI as master and example code  for Arduino as a node. I read the TMRh20 liinks and I dont found the solution. Node 1 appears as 05 on RPI, node 2 appears 04 and so on. Node 6 not work. Please help.