nRF24/RF24Mesh #76: Devduino doesn't receive a network address from Rpi master

aaddame, 2016-04-22:
I have an Rpi running as mesh master.
No other nodes in the network.
When I start a Devduino as first node, it doesn't receive an address. 
This sequence keeps going over and over again:
4293286141: MAC Received on 0 id 1 from 04444 to 0100 type 194
4293286141: MAC Sending to 04444 via 04444 on pipe 0
4293286141: NET Pipe 0 on node 04444 has address cccccc3ecc

When I start a nano as first node it receives an address.
4293381308: MAC Received on 0 id 1 from 04444 to 0100 type 194
4293381309: MAC Sending to 04444 via 04444 on pipe 0
4293381309: NET Pipe 0 on node 04444 has address cccccc3ecc
4293381366: MAC Received on 0 id 1 from 04444 to 00 type 195
4293381366 MSH: Rcv addr req from_id 2 
not allocated
not allocated
4293381368: NET Sending id 1 from 00 to 04444 type 128

They both are 328P processors, but the nano is running 16Mhz while the devduino runs at 1Mhz
Could this be an issue???

Attached is debug output from both situations
[debug.zip](https://github.com/TMRh20/RF24Mesh/files/231527/debug.zip)

Avamander, 2016-04-22:
Check the radio wiring. Do the examples of RF24 work?

aaddame, 2016-04-23:
Wiring? What about it. If you would have taken the time to visit the devduino website (http://www.seeedstudio.com/wiki/DevDuino_Sensor_Node_V2.2_(ATmega_328)) you would have noticed there is user wiring at all.

If I use static addresses the whole thing works.

Anyway , here is my analisys an solution.
Take a look at the trace I provided if you haven't done so already.
You'll see the devduino keeps sending NETWORK_POLL frames. This can only happen if it doesn't get NETWORK_POLL replies.
Before sending the frame, listening mode is turned off. Later in the code it is turned on again. But because of the huge speed difference between Rpi (16Mhz) and devduino ( 1Mhz in my case) , Rpi starts sending before the radio is in listening mode.

Solution: increase the delay in the following code fragment:
`if(header->type == NETWORK_POLL  ) {                                                            
          if( !(networkFlags & FLAG_NO_POLL) && node_address != 04444 ){                                
            header->to_node = header->from_node;                                                        
            header->from_node = node_address;                                                           
            delay(parent_pipe);                                                                                  
            write(header->to_node,USER_TX_TO_PHYSICAL_ADDRESS);                                         
          }                                                                                             
          continue;                                                                                     
        }     `

I used 10 in my case, but didn't experiment with it, so a lower delay may work as well

aaddame, 2016-04-23:
After running a while it fails again, because it drops into this code:
`if (!mesh.write(&data, 'M', sizeof(data))) {
    if ( ! mesh.checkConnection() ) {
      //refresh the network address
      mesh.renewAddress();
    }
  }`

I haven't determined yet how to fix.

Avamander, 2016-04-23:
Test if RF24's pingpair_dyn and Transfer work properly and have low packet loss.

aaddame, 2016-04-23:
Did something else.
I replaced mesh.update() at the start of the loop with 
`
if ( ! mesh.checkConnection() ) {
    //refresh the network address
    mesh.renewAddress();
  }
`

Because it has an implied network.update()
Seems to be running a lot better now

Avamander, 2016-04-23:
There's a reason for everything in the examples. Removing it is not a good
idea. Again, make sure you have original modules and all RF24's examples
work.

closed 2019-05-10 by TMRh20