nRF24/RF24Mesh #78: Problem with checkconnection

Fctoni, 2016-04-28:
Hello guys! This lib is awesome! Cheers to tmrh20 and avamander.

However, I got a little problem. Everything working nice, except when master node resets, it loses address table. Even then, nodes still get " true " on check connection function, so I cant renew adressess.

Shouldnt it return false, whenever master loses it's adress table?

Thank you!

Fctoni, 2016-04-28:
Let me explain a little better. While master is.turned off, check connection return false. As soon as it is powered.up, nodes get true on check connection . thanks

Avamander, 2016-04-28:
That is indeed something I noticed too, see https://github.com/TMRh20/RF24Mesh/issues/27, unfortunately it is not really possible to fix it easily, the way I solved it would be once in a while releasing the address and reconnecting to the mesh, that relists the node on the master and works fine.

closed 2019-05-10 by TMRh20