nRF24/RF24Mesh #79: Slave doesn't notice when connection with Master is lost

annawidera, 2016-04-30:
Hello!
To all involved: thank you very much for the libraries handling nrf24L01 communication. I know it requires very much effort to develop something like this. :) 

I encountered a problem with the **mesh examples**. Maybe I could debug it further in a proper way to give a further valuable feedback?

Let me describe briefly my case. 
I programmed Arduino Uno as a Master (code: **RF24Mesh_Example_Master**). 
I programmed 2 Bare Conductive Boards (based on Arduino Leonardo with ATmega32U4) as Slaves (code: **RF24Mesh_Example**). 

After a couple of seconds the Master stops receive any messages from one Slave. Then after a while from the second one. I suppose that the Slave board loses its connection with the Master. The Slave keeps sending messages but the Master doesn't receive them. 
The mesh.write() in the Slave returns true. I've moved also outside the write condition checking if the connection is still valid (mesh.checkConnection()). checkConnection also returns true. 

Slaves lose their connection with the Master in random order. Also, the time after which the connection is lost is random (I created a counter which counts every message received from any slave to check that). 

Power for the nrf24L01+ is taken from boards but with filtration (2.2uF electrolytic capacitor + two 10nF ceramic capacitors). 

Any advice will be highly appreciated. Maybe somebody could guide me into further debugging to determine if connection is really lost (but not recognized by the slave). 

Thank you!
Ania

annawidera, 2016-04-30:
I made further investigation. 

My current setup is: 
Bare Conductive (Arduino Leonardo) is the Master. 
I have also three slaves: two Arduino Uno's and one Bare Conductive.
Slaves are numbered as follow: 
Slave 1 = Arduino Uno, 
Slave 2 = Arduino Uno, 
Slave 3 = Bare Conductive. 

The Master stops printing messages from Slave 3 after about a 30 seconds. Then with only Slave 1 and Slave 2 communication stays stable. 

For that reason now I suppose that Slave 3 stops writing messages to mesh. But still mesh.write() returns true. (I have changed the topic of this issue)
Can you point me how to investigate that path further?
Thank you!
Ania

Avamander, 2016-04-30:
How are you verifying that mesh.write() returns true?

annawidera, 2016-05-01:
Hello Avamander! Thank you for your quick response. 

I am using the example: RF24Mesh_Example. The code for sending current millis() to the Master is as follows: 
`if (!mesh.write(&displayTimer, 'M', sizeof(displayTimer))) {
      Serial.println("Send error"); 
    } else {
      Serial.print("Send OK: "); Serial.println(displayTimer);
    }`

Slave 3 (Bare Conductive board) keeps logging "Send OK: value", even when Master does not receive any message from this slave.

Avamander, 2016-05-01:
Okay, then I'd recommend you check if you have original modules, I
encountered similar weird symptoms with my fake modules.

closed 2016-07-05 by Avamander