nRF24/RF24Mesh #82: rf24mesh_example and rf24mesh_example_master issue

grovermurray, 2016-05-29:
I appreciate any and all help as I am new to this library and to RF communication in general. I have ran most of the examples with no issues except for the mesh example/example master and the r24
getting started example. the rf24network rx and tx examples work great as well as the rf24 pingpair_dyn example. 

When I run the mesh example, the node only trys to renew the address and never successfully sends and data. The master node only returns a blank address list. 

I have scoured google and every other resource I could find, but I have not had any luck. The wiring appears to be correct (as the other examples work fine). I have replaced all of the libraries just in case something wasn't right. I read another issue where someone stated that the mesh library only worked with the dev version of the rf24network library so I used that with no luck. 

If anyone could shed any light on this, I would be most grateful.

other info:
I am using two arduino uno knock offs (did the same thing my one legit arduino uno) with two "what appears to be legit" nRF24L01+ modules. 
OS: Windows XP
Arduino IDE 1.6.9

Avamander, 2016-05-29:
This again sounds like fake nRF24s. Could you post the results of the
Transfer example?

grovermurray, 2016-05-29:
RF24/examples/Transfer/
*** PRESS 'T' to begin transmitting to the other node
Rate: 3.97
Payload Count: 124
Rate: 4.77
Payload Count: 149
Rate: 4.16
Payload Count: 130
Rate: 4.22
Payload Count: 132
Rate: 3.62
Payload Count: 113
Rate: 4.45
Payload Count: 139
Rate: 2.21
Payload Count: 69
Rate: 4.42
Payload Count: 138
Rate: 3.01
Payload Count: 94
Rate: 5.02
Payload Count: 157
Rate: 10.98
Payload Count: 343
Rate: 15.07
Payload Count: 471
Rate: 18.08
Payload Count: 565
Rate: 18.27
Payload Count: 571
Rate: 11.65
Payload Count: 364
Rate: 22.27
Payload Count: 696
Rate: 17.15
Payload Count: 536
Rate: 19.74
Payload Count: 617
Rate: 16.80
Payload Count: 525
Rate: 12.70
Payload Count: 397
Rate: 13.09
Payload Count: 409
Rate: 16.29
Payload Count: 509
Rate: 14.98
Payload Count: 468
Rate: 15.17
Payload Count: 474
Rate: 17.98
Payload Count: 562
Rate: 17.82
Payload Count: 557
Rate: 6.98
Payload Count: 218
Rate: 0.00
Payload Count: 0
Rate: 10.14
Payload Count: 317
Rate: 20.51
Payload Count: 641
Rate: 16.42
Payload Count: 513
Rate: 16.67
Payload Count: 521
Rate: 16.16
Payload Count: 505
Rate: 15.65
Payload Count: 489
Rate: 16.16
Payload Count: 505
Rate: 17.28
Payload Count: 540
Rate: 18.34
Payload Count: 573
Rate: 17.50
Payload Count: 547
Rate: 17.79
Payload Count: 556
Rate: 5.66
Payload Count: 177
Rate: 3.87
Payload Count: 121
Rate: 14.82
Payload Count: 463
Rate: 2.91
Payload Count: 91
Rate: 0.19
Payload Count: 6
Rate: 14.53
Payload Count: 454
Rate: 21.60
Payload Count: 675
Rate: 19.01
Payload Count: 594
Rate: 17.66
Payload Count: 552
Rate: 20.45
Payload Count: 639
Rate: 12.00
Payload Count: 375
Rate: 0.00
Payload Count: 0
Rate: 4.64
Payload Count: 145
Rate: 17.12
Payload Count: 535
Rate: 17.12
Payload Count: 535
Rate: 16.93
Payload Count: 529
Rate: 16.90
Payload Count: 528
Rate: 17.28
Payload Count: 540
Rate: 16.70
Payload Count: 522
Rate: 16.58
Payload Count: 518
Rate: 16.70
Payload Count: 522

I ran it very quickly as I have to head out of the house for a little while. The first time I ran it I had a rate of 120ish. It works bidirectionally as well.  Thank you for your assistance!

TMRh20, 2016-06-18:
I would assume a hardware, library or configuration/wiring issue. It should work with the libraries available through Arduino Library Manager and the RPi installer. See the [documentation](http://tmrh20.github.io/RF24/pages.html)

It is probably best to remove all RF24 related libraries via the IDE and manually, then reinstall per above. Then re-test with RF24 examples. If they (gettingstarted, gettingstarted call response, pingpair_dyn) are working then RF24Mesh should work. That is about all there is to it.

You can edit RF24Mesh_config.h and uncomment ```#define MESH_DEBUG``` to view some extra information, which may be helpful to post here.

closed 2016-07-05 by Avamander