nRF24/RF24Mesh #84: RF24Mesh::begin doesn't work in example, it seems

trademark18, 2016-06-07:
Hello.  I am running a Raspberry Pi 2 Model B with the RF24Gateway example.  I'm trying to follow the [video](https://www.youtube.com/watch?annotation_id=annotation_3313246909&feature=iv&src_vid=rBAIqAaRu0g&v=KZ5x1NF-Emc) TMRh20 put on YouTube a while back (apparently for an older version).  There are a couple differences in the ino file since the video was made, but I think it is close enough it should still work.  

I can run the RF24Gateway_ncursesInt program on the Pi and see my Arduino's radio pop up in the list if it is running one of the non-IP examples like RF24Mesh_Example.

**Problem**
When I run the InteractiveServer_Mesh sketch on my Arduino Nano (clone) I never get past the line where we mesh.begin();  This is odd because this call works perfectly well on several other sketches.  I can, it seems run basically any example except this one with little issue.  This particular example is the closest thing to my actual goal (see below) and I'd really like to get it working.

**Info**
RPi example: RF24Gateway_ncursesInt
Arduino example: InteractiveServer_Mesh <-- doesn't work
Arduino example: RF24Mesh_Example      <-- does work
Radios: nRF24L01+PA+LNA

**Goal** 
I'd like to get to the point where I can ping nodes on the mesh from the Pi.  From there I'd like to be able to ping the RF devices from other devices on my LAN (via the Pi).  Finally, I will dig into the mapping algorithm and implement a very efficient version of Breadth-Frist Search that I've developed that is specifically optimized for mesh networks.  I have the whole thing done in simulation, but now I'd like to reverse-engineer some of the work done here so I can implement my algorithm and see any differences (improvements?).

If I have left off anything important please let me know.  I typically don't post here in GitHub, so please bear with me.  Thanks!

TMRh20, 2016-06-07:
Cool, I assume you would need to set up a large number of nodes to test that kind of thing, so the results may be interesting. The problem sounds just like a [bug that just came up](https://github.com/TMRh20/RF24Mesh/issues/84) recently, and it should be fixed in [RF24Mesh v1.0.51](https://github.com/TMRh20/RF24Mesh/releases/tag/v1.0.51) which was just released. Only the Arduino device(s) should need updating, which you can do via the Arduino Library Manager.

trademark18, 2016-06-07:
Thanks for your quick reply.  I have 6 hardware radios completely built and ready to roll, so I'll be doing some testing shortly.  I downloaded the library a couple of months ago, so I'm sure I'm not on the recently released version you mentioned.  I'll try it again tonight and see how it goes.  Thanks!

trademark18, 2016-06-07:
It works! Thanks.

TMRh20, 2016-06-18:
https://github.com/TMRh20/RF24Ethernet/issues/15

closed 2016-06-18 by TMRh20