nRF24/RF24Mesh #87: I am unable to setup a dynamic mesh

rohinrohin, 2016-06-21:
**Setup:**
I have setup the master as from the example, (master to node msg) and have three nodes in the system as  

- 0-master
- 1-node1 
- 2-node2. 

I have setup the master to send a message to node2(2) every 5 seconds. 
It works like it should when the master is in range of node2. 

**The problem:**
When master is in range of node1 and node1 is in range od node2 (but node2 not in range of master). The master does not detect node2 at all. Is there anything I can do or am I doing something wrong?

Thanks

TMRh20, 2016-06-21:
In the same situation, does the master receive information if using the
rf24mesh_example_master rf24mesh_example sketches?

If you can monitor the end node using a laptop etc, please edit
RF24Mesh_config.h and uncomment #define MESH_DEBUG then reload Arduino &
please post the results.

rohinrohin, 2016-06-24:
Sorry for the delay. I will do that and send it to you asap.

rohinrohin, 2016-07-05:
Got it working, thanks!

closed 2016-07-05 by rohinrohin