nRF24/RF24Mesh #88: RF24mesh security

Toumi7, 2016-06-21:
Hello,
I'm doing a project that needs sensors connected with modules NRF24L01 + 2.4GHz and using this wonderful library (thank you ) 
I'd like to know if it is possible, through the library RF24Mesh (RF24 and RF24Network), have a secure connection  (an authentication or encryption algorithm).
Thanks for the answer in advance

TMRh20, 2016-06-21:
Yes and no?

1. With RPi/Linux devices one can literally just use TCP/IP and
SSH,HTTPS,etc.
2. I would suggest similar concepts for other devices that have support for
authentication or encryption protocols (ESP8266 and such) and handle
authentication and encryption high up on the stack as per the OSI model,
instead of trying to encrypt and protect the network like with WiFi.
3. Real security on AVR devices without additional hardware is debatable I
think, and I'm not so sure about what level of security would be achieved.

Toumi7, 2016-06-21:
does that mean that anyone who has this library and a NRF24 module can listen to all messages and simply hack any node...

TMRh20, 2016-06-21:
Just like any unencrypted email, chat, etc you send across the Internet, anybody with access to the network can sniff traffic. I'm not sure how someone would 'hack' a temperature sensor, but I'll leave the hacking part to imagination.

ppotok, 2016-06-21:
If I may to interrupt conversation,
I think "hack" mean when you remote outlets, and someone in range can
remote my things its not good.

___ ___ ___

S pozdravem - Petr Potoček
Sincerely - Petr Potoček
petr.potocek.cz
petr@potocek.cz / p.potok@gmail.com

<p.potok@gmail.com>
----------------------------------------------------------

Toumi7, 2016-06-22:
yes @ppotok  that's what i mean exactly 
@TMRh20  so to sniff traffic, 'he' only need an NRF24 module and this librarie or he must be a child of my network??

Avamander, 2016-06-22:
One can just listen to all packets on all pipes and channel if one wants to
do that. There are a lot of ways to make this thing more secure, RF24* does
not offer you any security and is not intended to do so.

Toumi7, 2016-06-22:
thanks @Avamander 
so how can i get a mesh netWork  with some security or encryption and using arduino ( i want to use arduino nano due their size )

ppotok, 2016-06-22:
And what about to use some kind of struct in payload of different types
etc, and there set coded name&password ? Attacker can sniff payload, but it
takes time to decipher struct types...

Avamander, 2016-06-22:
That's not security. Use some real encryption or signing algorithms.

Toumi7, 2016-06-23:
@Avamander 
even if i use an AES encryption, that will be only for the payload and the header
there will be a problem if the rest of the frame (destination and sender address ... ) will be visible for all ?

Avamander, 2016-06-24:
@Toumi7 An device like Arduino can only implement basic, not very strong, encryption algorithms, you should look towards signing or hardware encryption chips.

spaniakos, 2016-06-24:
Last year i made a framework tonimplement security.
The framework is almost done but i had to stop as i had to serve in the
military. After the end of august i will update according to new additions
and post it to the repository. If you look at my repos i have encryption
algorythms. What is miasing the framework on top of the library, that i
have locally in my computer. @TMRh20 might remember that initiative. All i
can tell you is the framework implements mechanisms in orser to to login a
node to the network. Key handing, and encrypted payloads. Encryprion can be
point to point.
It worked with 99% success rate ( i have some packet loaa in the proccess
of re-key )
I hope i can post the framework back to the community by the end of august
or earlier.

Toumi7, 2016-06-24:
@spaniakos thanks you 
but honestly i'm not that good concerning security details ... i know what is encrypting and i know some algorithms (AES DES ..) but i learned by my self so i can say that i'm understanding only  70-80% of all that .. 
(exactly i didn't understand what is
 - key handing
- point to point encryption 
-re key ??

Scubatcus, 2016-10-03:
What I did to make it secure in a very simple way is XOR the payloads with a 16 bits key.
The Raspberry (Master) sends a new key every time it communicates with the Arduino so this makes it very difficult to crack. And even when that particular payload was decoded successfully, it's only valid for that specific payload and cannot be used again.

Besides, all my payloads have their own timestamp, CRC and sequence number so things like 'replaying' a payload a tampering with it won't work either.

Just imagine I'm your neighbour and capture the payload for switching on/off outdoor lights. Now if you did not take counter measures I could easy switch your light on and off any time I like to. So beware of replaying payloads.

There is however a problem with the Mesh. I figured that when there is a Node with causes problems, in my case a NRF power problem, the master goes nuts, receiving payloads from a node with -1. As soon as I take the error node out of the network the others can communicate normally again.

Another thing I ran into is that sometimes I really need physically power off/on the NRF on the Raspberry. I'm planning to power it from a GPIO pin instead of directly connecting to 3v3.

I do the same thing on battery controlled Arduino's, Powering the module (&sensors) from a digital pin so when WDT kicks in, the NRF & Sensors are also power cycled. The powerdown function of the NRF itself is just not enough.

Avamander, 2016-10-03:
One way is to use a hardware encryption chips that has SPI or I2C bus, I am
using ATsha204s but those don't exactly do encryption.

SafwatHalaby, 2018-06-17:
@Toumi7 Microcontrollers are too slow for safe crypgrography. If security is important,Many Zigbees (such as the Xbee) support hardware-based AES encryption, and that may be a better choice than NRF24.

@Scubatcus , if you are doing this for a minor project, then this simple security layer is fine.

However, *do NOT* use it for anything serious. Your algorithm is basically "security through simple obscurity" and is trivially crackable by any person with basic cryptography knowledge.

Truly secure cryptography is surprisingly hard. Even experts can get it wrong. Please search for the term "never roll your own crypto".

Avamander, 2018-06-17:
> Microcontrollers are too slow for safe crypgrography.

Fast enough if you only want to send one payload once in a while.

SafwatHalaby, 2018-06-17:
> Fast enough if you only want to send one payload once in a while.

I agree. I've quickly removed that sentence after posting the message.

igormp, 2018-10-11:
A thing that I've been toying with is encryption + authentication through software, so the final payload that goes into the mesh is somewhat secure and transparent to the actual network.

One way to achieve that is by using something like AES, which has many implementations out there for many microcontrollers, along with something like [CBC-MAC](https://en.wikipedia.org/wiki/CBC-MAC) or [CMAC](https://en.wikipedia.org/wiki/One-key_MAC) for authentication (or even do [CCM mode](https://en.wikipedia.org/wiki/CCM_mode), which describes how to do both things at once).

In my case, I modified [AESLib](https://github.com/DavyLandman/AESLib) a bit and made my own [auth lib](https://github.com/igormp/CMACLib) (still a WIP) on top of it to achieve something that's both encrypted, meaning that no one can read it's contents without having the secret key, and authenticated, securing my network against replay attacks to some degree (which is the most important thing in a network of actuators IMO).

Having an option to have this built-in into the library would be nice, but would probably require a lot of work.

closed 2019-05-10 by TMRh20