nRF24/RF24Mesh #89: ESP8266 Support

rohinrohin, 2016-06-24:
I have setup the module as http://d.av.id.au/blog/esp8266-hardware-spi-hspi-general-info-and-pinout/ and had used the code in https://gist.github.com/rohinrohin/7a1ca4f4cec1d4c93480945b988cbe1c. The module still boot-loops as described in https://gist.github.com/crcastle/6bab21445c857993d1d8. I saw an issue raised on RF24 and was fixed. I don't know why the esp is still crashing?

Thanks. 
PS: in the gist, I uploaded commented and uncommented to the esp and the device still reboots. Referencing #125 from RF24.

TMRh20, 2016-06-28:
I have typically been using the git version of
https://github.com/esp8266/Arduino with Arduino IDE 1.6.9

It should work with the standard RF24 examples providing the pins and
connections are correct. Note: My nodeMCU uses GPIO numbers rather than the
digital pin numbers written on the board. ie: RF24 radio(2,15) or (4,15)
work for me

rohinrohin, 2016-06-28:
Okay awesome, I will test again from the latest version and get back to you. I will close the issue for now. 
Thanks.

closed 2016-06-28 by rohinrohin