nRF24/RF24Mesh #91: RF24 examples works, RF24Mesh example's - can't connect to master...

lpetrov, 2016-07-02:
I'm wondering..
What can be the issue of this not working?
In my master node (Running ..Example_Master) i can see that the node1 is getting address 5, but nothing else is being received.
On the client node 1 - RF24Mesh_Example, i can see the node is trying to connect...
<pre>
Connecting to the mesh...
22 MSH: Poll 
28 MSH: Poll > -64dbm 
82 MSH: Poll OK 
82 MSH: Got poll from level 0 count 1
84 MSH: Req addr from 0
94 Set address:    5
496 MSH: Poll 
555 MSH: No poll from level 1
613 MSH: Poll 
671 MSH: No poll from level 2
739 MSH: Poll 
798 MSH: No poll from level 3
880 MSH: Poll 
883 MSH: Poll > -64dbm 
939 MSH: Poll OK 
939 MSH: Got poll from level 0 count 1
941 MSH: Req addr from 0
950 Set address:    5
1361 MSH: Poll 
...and this goes on and on and on and it never connects...
</pre>

Its weird, because RF24 examples worked out perfectly with both of my Long/regular range modules.
So i don't see why this is not working.
Is there any extra-debugging trick that i can do to see whats going on/provide more information for this issue?

Thanks!
L

ddlsmurf, 2016-07-03:
I have the same issue with 1.1.6. Pingback works boths ways, but mesh connection only works between AVRs. This issue appears when the master or client is an ESP8266 (though fair sure it used to work).

lpetrov, 2016-07-03:
I'd tried to use 2 nanos or a nano and an 2560 here...So not sure if it is ESP8266 related...

ddlsmurf, 2016-07-03:
For my part I've been running it in all combinations with a Teensy 3.1 and ATmega328P, if the ESP is out of the loop it works flawlessly. On the ESP, RFNetwork has rather severe packet loss but on the whole gets the packets through, but RFMesh barely ever even start talking. Could this be timing related ? (the ESP hi-jacks delay() to run its background stuff) Or perhaps Ipetrov and I have the same hardware error ?

lpetrov, 2016-07-03:
You may be having a different problem... In my case I'd tested with 2 nanos and 1 nano and a mega2560

lpetrov, 2016-07-03:
(^^ and RF24Mesh did not worked, however RF24 getting started example did worked perfectly)
Today i also tried to power the nrfs from 2 aa 1.5v batteries (~2.5-2.8v left) and still - the same problem :/

ddlsmurf, 2016-07-03:
For my part it seems like it was a faulty nrf24l01+ module - replacing it solved everything.

TMRh20, 2016-07-04:
```
950 Set address:    5
1361 MSH: Poll 
```
This line seems to indicate that the device is probably getting through address renewal, but the final confirmation is failing. This is just a standard network write to the master with confirmation.

I would suggest making sure all your RF24 libs are up to date, but other than that, I would look towards testing with RF24Network examples to see if that is working normally. If so, then it might be something specific to RF24Mesh, but I would also think maybe hardware related.

TMRh20, 2016-07-04:
Also, I've noticed some lag/delays with ESP/NodeMCU that aren't present with AVR devices, but they do seem to work. I think it has to do with the software driven IP stack etc.

madias123, 2016-07-06:
I confirm that the Mesh-examples isn't working (2 Unos's)
I'm using the same hardware setting for all examples.
NRF24: everything is working
NRF24-network: everything is working
But I can't get the mesh-demo working. (No connection between master and slave - on slave only the message "Connecting to the mesh...", on master the Assigned Addresses list is blank.)

Avamander, 2016-07-06:
This is commonly caused by fake modules, switch to some other modules if
you can. You can also try decoupling but I doubt it'll fix the issues.

lpetrov, 2016-07-06:
Meanwhile I agree that there is always a chance that all 4 of my batches are fakes...I really feel that this does not make sense, since I ordered those all from users with tons of orders and 5 out of 5 stars rating.
However, because of this whole fake-thing I'm rethinking the design of my home automation project and would move to something more stable then NRFs.
So thanks for the help anyway, feel free to close this issue in case you think that would not be relevant/helpful for other users.

madias123, 2016-07-06:
Not all fakes are "bad": The Si24R1 has a problem with ACK bit inverted, some "epoxy blop" modules are missing components, but the rest (like I use) are still ok,  uses maybe(!) more power, less range but are fully functional!
As I wrote before, I ONLY have problems with the mesh-examples nor with the NR24 or the NR24 network library. They run all out of the box. (with good signal). I'm also using the additional power module for each NRF24.

Avamander, 2016-07-06:
@lpetrov @madias123  Please run the transfer example of RF24 and report the packet loss and data rate.

The thing with fakes are that they fail just like you've described, sometimes they work sometimes they don't. I too ended up with a bunch of fake modules, a few batches of them. First of all, mixed environments of different fake modules cause issues with RF24Mesh (own experience shows the same symptoms you have). Then the fakes also have stability issues, packet loss, wrongly or missing features, they are a PITA.

For those who are testing these modules out or trying to determine if they're fake.

This is how I tested them:

1) I made a table (and tested every combination) for basically any example. The best I think was Transfer, it highlighted the minor differences the most. For ex. packet loss with original ones stays 0 or 1 usually, but with fakes >20 is common. When a capacitor is attached the fake ones seem to work better. 

For ex:

Transfer R: | Module 1 | Module 2 (fake)
------------ | ------------- | -------------
Module 1 | 47.09 KB/s | 35.95 KB/s
Module 2 (fake) | 38.42 KB/s | 30.23 KB/s

You can see the fakes suffer from lower data rates too.

2) Look at the writing of the NRF chip. If the text is like on the right one, it's genuine most likely, if it is like the left one, it is fake.

[image]

3) If the chip is under epoxy blob, it is fake.

I think if you buy genuine models on eBay I would love to create a list of sellers/listings who sell genuine so that there would be less headache. As the faults appear after further testing the eBay ratings do not show that.

madias123, 2016-07-13:
Dear Avamander,
it's a pain to get legit NRF24's on Ebay and/or Aliexpress. Reason: 99% are counterfeits. Even if you are lucky and get genuinous ones from one seller, there could be the risk that you get fake ones after one month.
Some "funny" stuff:
Aliexpress, search term nrf24 + original (and/or) genuine you got results like this:
http://www.aliexpress.com/item/Free-Shipping-50PCS-LOT-NRF24L01-NRF24L01P-Wireless-Transceiver-Module-NRF24L01-2-4GHz-For-AVR-ARM-Ar/32552562528.html
Even on the photo you can see it's a fake.

Avamander, 2016-07-13:
@madias123 I did not claim otherwise. Just don't pick the cheapest offer and order two before ordering 10.

mamama1, 2017-01-19:
Hi

i recently ordered these, and they seem to work fine. At least they compare to the sparkfun modules which I think use original NRFs as they cost about $20 each.

https://www.aliexpress.com/item/Wireless-Module-NRF24L01-original-2-4-Ghz-RF-transceiver-SPI/32712174920.html?spm=2114.13010608.0.0.vZu4S3

https://www.aliexpress.com/item/Free-shipping-BAITE-BRAND-10PCS-nRF24L01-Wireless-Module-2-4G-Wireless-Communication-Module-Upgrade-Module/707309324.html?spm=2114.13010608.0.0.vZu4S3

no issues whatsoever. on my known-fake modules I have to spmetimes lay a finger on the board/crystal/ic/antenna and/or solder a 100nf ceramic capacitor directly to VCC and ground. none of these was necessary on the linked modules to make them work.
i was even sending 320 byte of payload (fragmented) in a loop without a problem.

Avamander, 2017-01-20:
You should try the 20$ and AliExpress ones in the same mesh. Clones +
originals tend to not work with each other. In addition to that clones with
originals in all the three unique cominations all give different results
with the Transfer example.

mamama1, 2017-01-20:
I did, everything works.
Basically I have 4 batches of modules:
- Sparkfun 20$/piece - work 100%
- cheap generic modules 0,7$/piece - work most of the time, need ceramic capacitor + sometimes a finger on crystal/ic/antenna
- BAITE, linked above, 1,65$/piece - work 100% without modifications
- Robotdyn, linked above, 1,99$/piece - work 100% without modifications

the NRF ICs on the BAITE and Robotdyn modules look like the one on the Sparkfun 20$ module. On the other hand, the ICs on the cheap, generic 70c modules look completely different.

All modules can communicate with all of the modules so I guess the 70c modules I got aren't too bad, at least they seem to have everything implemented correctly. The cheap 70c modules need a capacitor and/or sometimes a finger on ic/crystal/board/antenna to work. And then, they still only work in 90% of the time.

I just wanted to share my findings about good NRF modules, as it isn't too easy to get working NRFs on aliexpress. The two modules linked above work perfectly.

closed 2019-05-10 by TMRh20