nRF24/RF24Mesh #95: "Automatic" relaying

Ryan00793, 2016-08-20:
An explicit statement on the front page of your doxygen for this project that says something about needing to setChild(true) to get _automatic_ relaying to work

>   - Automatic routing and handling of data - Nodes will join together to extend the range of radio links by routing data as required.

TMRh20, 2016-08-22:
See the [setChild()](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#ae183c358915a1b7fbf1b19db8ba66a16) function 

Routing is automatic unless manually disabled.

Ryan00793, 2016-08-27:
The actual routing part works fine, however to extend the range of the radios, I must call that function.

TMRh20, 2016-08-29:
This functionality is controlled by the RF24Network networkFlags variable
which is initialized [here](
https://github.com/TMRh20/RF24Network/blob/master/RF24Network.cpp#L64)

If you are running custom code, there might be a memory issue causing that
variable to be modified. Does this happen with the default examples?

closed 2019-05-10 by TMRh20