nRF24/RF24Mesh #96: Support for NRF24LE1 or NRF24LU1+.

rajithavk, 2016-09-19:
Can this stack be built for NRF24LE1 or NRF24LU1+ ? Both these SoC's have included their own  MCU. If this Mesh stack can be used on them, it's going to be one powerful module since no other MCUs are needed.

TMRh20, 2016-09-19:
It almost certainly could be.

The radio API seems to be much the same as 24L01+ , so I assume it would be a matter of adapting the core RF24 driver via the [portability template](http://tmrh20.github.io/RF24/Portability.html) or if there is an Arduino core available, it would just work.

Above that, there should be no other dependencies.

rajithavk, 2016-09-22:
Great, I didn't know about the portability template. Thanks a lot 👍.
BTW, my real requirement is a sensor network so, that time synchronization is very critical. Does the current implementation has built in time synchronization mechanism? or will i have to work on it?
Thank you.

TMRh20, 2016-09-22:
Nothing really built in regarding time sync. Nodes can be set up to request the time from a single device with RTC capabilities via standard communication, as in the RF24Gateway examples.

closed 2019-05-10 by TMRh20