nRF24/RF24Mesh #97: network.peek always returns 0

smiron, 2016-09-19:
Hi,

In my setup I have a master and 2 nodes.
The first node sends data to the second node when a condition occurs. However, the sent data size is dynamic. So the second node needs to know how much memory to allocate and read from the network.

I saw in one of the example projects that you use the return value from network.peek as the size of the data to be received.
https://github.com/TMRh20/RF24Mesh/blob/master/examples/RF24Mesh_Example_Node2NodeExtra/RF24Mesh_Example_Node2NodeExtra.ino

However in my setup network.peek always returns 0.

PS: sending and receiving data using a header with a known size upfront works perfectly.
PPS: I tried the example above but I couldn't get it to work. The nodes don't communicate.

Is this pattern still supported?

Thank you,
Sorin

TMRh20, 2016-09-20:
Yup, it works. Due to the rapid communication & node lookups involved in this example, the modules need to be performing optimally.

Testing with the RF24 pingpair_dyn example will tell you if your module are working with dynamic payloads. If so, adding network.peek() to the RF24Network helloworld_rx could indicate a coding problem if it returns 0.

TMRh20, 2016-09-20:
FYI it turns out this [was a bug](https://github.com/TMRh20/RF24Network/commit/3fe2560f8d22cfbdeed52e327eb76524aff528b4) that was fixed, but was not put out as a 'release' so it was not available via Arduino Library Manager. I just released the current code with other updates and realized it, so think it should be available (v1.0.7) within a matter of hours.

smiron, 2016-09-20:
Wow. Very speedy response and fix :).
Thank you very much! It works perfectly now :D
I'm not using ready made modules. I made my own.

Thank you very much again :)

closed 2016-09-20 by smiron