nRF24/RF24Mesh #99: mesh.begin() in RF24Mesh_Example always return false

SachithW, 2016-09-28:
Hi, I am playing with RF24mesh for a while. The RF24Mesh_Example,ino file doesn't work for me. I debuged it as follows.

 `   bool b = mesh.begin();
  Serial.print("mesh.begin()");
  Serial.println(b);` 

mesh.begin() always return zero for me. But RF24Mesh_Example_Master.ino works fine and it return true.

I am using nrf24Lo+ modules and I guess those are fake.

Do you have any idea whats happening???

Thank You.

TMRh20, 2016-10-08:
I would suggest testing with the core RF24 examples, and ensure they are working up to par. If it returns false, the device is not communicating properly (cannot get an address from master)

SachithW, 2016-10-11:
Thanks TMRh20 for your suggestion. There was a power issue in the circuit. This problem was solved when I add a cap to the power IN.

closed 2016-10-11 by SachithW