nRF24/RF24Mesh #100: Mesh Connection Problem

cagdaseckin, 2016-10-13:
I'm using NRF24L01 to make mesh network. [This example](http://starter-kit.nettigo.eu/2014/connecting-and-programming-nrf24l01-with-arduino-and-other-boards/) works fine for peer to peer communication. However RF24Mesh library didint. Always giving these messages:

"RF24Mesh_Example" code gives only :

Renewing Address
"RF24Mesh_Example_Master" code gives only :

0

********Assigned Addresses********
**********************************

********Assigned Addresses********
**********************************

Avamander, 2016-10-13:
In order to make sure it's the library make sure to do all these steps:

1) If the nRF24L01 has a blob not a chip, it's a clone and you should get
rid of them

2) Check the wiring, use RF24's pingpair to test it

3) Delete Arduino IDE and all libraries then redownload the latest version
of the IDE and from that the library

If the situation does not fix itself after any of the steps post the output
of Pingpair and printDetails():

cagdaseckin, 2016-10-14:
First of all thanks for rapid response.

1- The nrf24l01 has chip
2- Pingpair test gives these results:

**Receiver:**

```
RF24/tests/pingpair_test/

ROLE: Receiver

+READY press any key to start

Configuration	 = 2

STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xccccccccc3 0xe8e8f0f0e1
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xcccccc3ecc
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3e
RF_CH		 = 0x0a
RF_SETUP	 = 0x27
CONFIG		 = 0x0b
DYNPD/FEATURE	 = 0x3f 0x06
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MAX

+OK Recv size=4 val=ABCD len=4

Ack Payload:Sent

Recv size=6 val=ABCDEF len=6

Ack Payload:Sent

Recv size=8 val=ABCDEFGH len=8

Recv size=10 val=ABCDEFGHIJ len=10

Ack Payload:Sent

Recv size=12 val=ABCDEFGHIJKL len=12

Ack Payload:Sent

Recv size=14 val=ABCDEFGHIJKLMN len=14

Ack Payload:Sent

Recv size=16 val=ABCDEFGHIJKLMNOP len=16

Ack Payload:Sent

Recv size=18 val=ABCDEFGHIJKLMNOPQR len=18

Ack Payload:Sent

Recv size=20 val=ABCDEFGHIJKLMNOPQRST len=20

Ack Payload:Sent

Recv size=22 val=ABCDEFGHIJKLMNOPQRSTUV len=22

Ack Payload:Sent

Recv size=24 val=ABCDEFGHIJKLMNOPQRSTUVWX len=24

Ack Payload:Sent

Recv size=26 val=ABCDEFGHIJKLMNOPQRSTUVWXYZ len=26

Ack Payload:Sent

Recv size=28 val=ABCDEFGHIJKLMNOPQRSTUVWXYZ78 len=28

Ack Payload:Sent

Recv size=30 val=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890 len=30

Ack Payload:Sent

Recv size=32 val=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012 len=32

Ack Payload:Sent

Recv size=4 val=ABCD len=4

Ack Payload:Sent

Recv size=6 val=ABCDEF len=6

Ack Payload:Sent

Recv size=8 val=Recv size=8 val= men=;

RF24/tests/pingpair_test/

ROLE: Receiver

+READY press any key to start
```

**Sender** 

```
RF24/tests/pingpair_test/

ROLE: Sender

+READY press any key to start

Configuration	 = 2

STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe8e8f0f0e1 0xe8e8f0f000
RX_ADDR_P2-5	 = 0x33 0xce 0x3e 0xe3
TX_ADDR		 = 0xe8e8f0f0e1
RX_PW_P0-6	 = 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		 = 0x3e
EN_RXADDR	 = 0x3f
RF_CH		 = 0x0a
RF_SETUP	 = 0x27
CONFIG		 = 0x0a
DYNPD/FEATURE	 = 0x3f 0x06
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 8 bits
PA Power	 = PA_MAX

Now sending length 4...Send:OK 

Now sending length 6...Send:OK 

Now sending length 8...Send:OK 

Now sending length 10...Send:OK 

Now sending length 12...Send:OK 

Now sending length 14...Send:OK 

Now sending length 16...Send:OK 

Now sending length 18...Send:OK 

Now sending length 20...Send:OK 

Now sending length 22...Send:OK 

Now sending length 24...Send:OK 

Now sending length 26...Send:OK 

Now sending length 28...Send:OK 

Now sending length 30...Send:OK 

Now sending length 32...Send:OK 

Now sending length 4...Send:OK 

Now sending length 6...Send:OK 

Now sending length 8...Send:OK 

Now sending length 10...Send:OK 

Now sending length 12...Send:OK 

Now sending length 14...Send:OK 

Now sending length 16...Send:OK 

Now sending length 18...Send:OK 

Now sending length 20...Send:OK 

Now sending length 22...Send:OK 

Now sending length 24...Send:OK 

Now sending length 26...Send:OK 

Now sending length 28...Send:OK 

Now sending length 30...Send:OK 

Now sending length 32...Send:OK 

Now sending length 4...Send:OK 

Now sending length 6...Send:OK 

Now sending length 8...Send:OK 

Now sending length 10...Send:OK 

Now sending length 12...Send:OK 

Now sending length 14...Send:OK 

Now sending length 16...Send:OK 

Now sending length 18...Send:OK 

Now sending length 20...Send:OK 

Now sending length 22...Send:OK 

Now sending length 24...Send:OK 

Now sending length 26...Send:OK 

Now sending length 28...Send:OK 

Now sending length 30...Send:OK 

Now sending length 32...Send:OK 

Now sending length 4...Send:OK 

Now sending length 6...Send:OK 

Now sending length 8...Send:OK 

Now sending length 10...Send:OK 

Now sending length 12...Send:OK 

Now sending length 14...Send:OK 

Now sending length 16...Send:OK 

Now sending length 18...Send:OK 

Now sending length 20...Send:OK 

Now sending length 22...Send:OK 

Now sending length 24...Send:OK 

Now sending length 26...Send:OK 

Now sending length 28...Send:OK 

Now sending length 30...Send:OK 

Now sending length 32...Send:OK 

Now sending length 4...Send:OK 

Now sending length 6...Send:OK 

Now sending length 8...Send:OK 

Now sending length 10...Send:OK 

Now sending length 12...Send:OK 

Now sending length 14...No responses received.  Are interrupts connected??

Send:OK 

+OK FAIL
```

3- All libraries was deleted and reinstalled

Still not working mesh

Avamander, 2016-10-14:
Please post the output of pingpair_dyn from /RF24/examples/ not
/RF24/tests/.

cagdaseckin, 2016-10-14:
```
RF24/examples/pingpair_dyn/
ROLE: Ping out
Now sending length 4
Got response size=4 value=ABCD
Now sending length 5
Got response size=5 value=ABCDE
Now sending length 6
Failed, response timed out.
Now sending length 7
Got response size=7 value=ABCDEFG
Now sending length 8
Got response size=8 value=ABCDEFGH
Now sending length 9
Got response size=9 value=ABCDEFGHI
Now sending length 10
Got response size=10 value=ABCDEFGHIJ
Now sending length 11
Got response size=11 value=ABCDEFGHIJK
Now sending length 12
Got response size=12 value=ABCDEFGHIJKL
Now sending length 13
Got response size=13 value=ABCDEFGHIJKLM
Now sending length 14
Got response size=14 value=ABCDEFGHIJKLMN
Now sending length 15
Got response size=15 value=ABCDEFGHIJKLMNO
Now sending length 16
Got response size=16 value=ABCDEFGHIJKLMNOP
Now sending length 17
Got response size=17 value=ABCDEFGHIJKLMNOPQ
Now sending length 18
Failed, response timed out.
Now sending length 19
Failed, response timed out.
Now sending length 20
Failed, response timed out.
Now sending length 21
Failed, response timed out.
Now sending length 22
Got response size=22 value=ABCDEFGHIJKLMNOPQRSTUV
Now sending length 23
Got response size=23 value=ABCDEFGHIJKLMNOPQRSTUVW
Now sending length 24
Got response size=24 value=ABCDEFGHIJKLMNOPQRSTUVWX
Now sending length 25
Got response size=25 value=ABCDEFGHIJKLMNOPQRSTUVWXY
Now sending length 26
Got response size=26 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ
Now sending length 27
Got response size=27 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7
Now sending length 28
Got response size=28 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78
Now sending length 29
Got response size=29 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789
Now sending length 30
Got response size=30 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890
Now sending length 31
Got response size=31 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78901
Now sending length 32
Got response size=32 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012
```

```
RF24/examples/pingpair_dyn/
ROLE: Pong back
Got response size=22 value=ABCDEFGHIJKLMNOPQRSTUV
Sent response.
Got response size=23 value=ABCDEFGHIJKLMNOPQRSTUVW
Sent response.
Got response size=24 value=ABCDEFGHIJKLMNOPQRSTUVWX
Sent response.
Got response size=25 value=ABCDEFGHIJKLMNOPQRSTUVWXY
Sent response.
Got response size=26 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ
Sent response.
Got response size=27 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7
Sent response.
Got response size=28 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78
Sent response.
Got response size=29 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789
Sent response.
Got response size=30 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890
Sent response.
Got response size=31 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78901
Sent response.
Got response size=32 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012
Sent response.
Got response size=4 value=ABCD
Sent response.
Got response size=5 value=ABCDE
Sent response.
Got response size=6 value=ABCDEF
Sent response.
Got response size=7 value=ABCDEFG
Sent response.
Got response size=8 value=ABCDEFGH
Sent response.
Got response size=9 value=ABCDEFGHI
Sent response.
Got response size=10 value=ABCDEFGHIJ
Sent response.
Got response size=11 value=ABCDEFGHIJK
Sent response.
Got response size=12 value=ABCDEFGHIJKL
Sent response.
Got response size=13 value=ABCDEFGHIJKLM
Sent response.
Got response size=14 value=ABCDEFGHIJKLMN
Sent response.
Got response size=15 value=ABCDEFGHIJKLMNO
Sent response.
Got response size=16 value=ABCDEFGHIJKLMNOP
Sent response.
Got response size=17 value=ABCDEFGHIJKLMNOPQ
Sent response.
Got response size=18 value=ABCDEFGHIJKLMNOPQR
Sent response.
Got response size=19 value=ABCDEFGHIJKLMNOPQRS
Sent response.
Got response size=20 value=ABCDEFGHIJKLMNOPQRST
Sent response.
Got response size=21 value=ABCDEFGHIJKLMNOPQRSTU
Sent response.
Got response size=22 value=ABCDEFGHIJKLMNOPQRSTUV
Sent response.
Got response size=23 value=ABCDEFGHIJKLMNOPQRSTUVW
Sent response.
Got response size=24 value=ABCDEFGHIJKLMNOPQRSTUVWX
Sent response.
Got response size=25 value=ABCDEFGHIJKLMNOPQRSTUVWXY
Sent response.
Got response size=26 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ
Sent response.
Got response size=27 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7
Sent response.
Got response size=28 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78
Sent response.
Got response size=29 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789
Sent response.
Got response size=30 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ7890
Sent response.
Got response size=31 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ78901
Sent response.
Got response size=32 value=ABCDEFGHIJKLMNOPQRSTUVWXYZ789012
```

Avamander, 2016-10-14:
Dynamic payloads seem to be supported it seems. Now what does the Transfer
example show?

cagdaseckin, 2016-10-14:
```

RF24/examples/Transfer/
*** PRESS 'T' to begin transmitting to the other node
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.03
Payload Count: 1
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.03
Payload Count: 1
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.03
Payload Count: 1
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
```
```

RF24/examples/Transfer/
*** PRESS 'T' to begin transmitting to the other node
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
Rate: 0.00
Payload Count: 0
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Initiating Basic Data Transfer

```

cagdaseckin, 2016-10-14:
I made connections with solder from the beginning. Once, Mesh connection worked for a while (2 min). than I want to generate a new code from mesh and upload. Now, and now it is again not working with examples. **There is no change in libraries and example files.**

Avamander, 2016-10-14:
Solder 1000uF electrolytic capacitors to the modules (3.3V and GND pin). By
default original modules should not require them, but some clone's
performance improves tremendrously.

acseckin, 2016-12-04:
A wire soldering to antenna is a solution for us. Now works fine.

closed 2016-12-04 by acseckin